lgsvl/simulator #839: Noisy point cloud recovered from depth image?

Doodle1106, 2020-07-06:
Hi,

I tried to recover the point cloud from the depth camera and the result is noisy along the object edges.

Below is the depth image and recovered point cloud:

[image]
[image]

following is a side view of the point cloud:
[image]

code to reproduce the result:
```

import cv2 as cv
import numpy as np
# from pyntcloud import PyntCloud
import pandas as pd
import open3d as o3d

def depth_to_p3ds(depth_img, fx, fy, cx, cy):
  assert(depth_img.size)
  
  h = depth_img.shape[0]
  w = depth_img.shape[1]
   
  p3ds = []
  for v in range(h):
    for u in range(w):
      z = depth_img[v][u]
      
      if z==0:
        continue
      
      x = z * (u - cx) / fx
      y = z * (v - cy) / fy

       # open3d and opencv has opposite Y direction
      y = - y 
      
      p3ds.append([x, y, z])

  return p3ds

def visualize_pointcloud(p3ds):
    pcd = o3d.geometry.PointCloud()
    pcd.points = o3d.utility.Vector3dVector(p3ds)
    o3d.visualization.draw_geometries([pcd])

depth_img = cv.imread("depth.png")
depth_img = cv.cvtColor(depth_img, cv.COLOR_BGR2GRAY)
pc = depth_to_p3ds_open3d(depth_img, 550.0394833255903, 550.0394833255903, 376, 240)
visualize_pointcloud(pc)
```

EricBoiseLGSVL, 2020-07-06:
@rongguodong Any idea what could be the issue?

closed 2020-07-13 by Doodle1106

Rwik2000, 2021-04-18:
Anyone has any idea? I am facing the same problem. @EricBoiseLGSVL have you resolved it?