lgsvl/simulator #840: Can not create waypoint.

wangludong, 2020-07-06:
Hi
I'd like to annotate a new map.
According to the manual here [https://www.lgsvlsimulator.com/docs/map-annotation/](url),
I try it with following steps:

1.Create a empty object and rename it to Map, then drag a MapHolder script to it.
[image]

2.Create a empty object under Map and rename it to TrafficLanes, then drag it to the Traffic Lanes holder.
[image]

3.Open Annotate HD Map and drag TrafficLanes to Parent Object, then press Lane Section.
[image]

The log is out there but i get nothing under TrafficLanes.
Is there something wrong with my operation?

EricBoiseLGSVL, 2020-07-06:
@wangludong 
- You might have the existing MapHolder for SF causing issues.  You should make a new scene to create new annotations.
- Don't mix multiple map meshes and annotations
- You should reset the transforms for the holder objects.  They need to be 0,0,0 pos and rot with scale 1, 1, 1
- If you make a new scene and save it, and have no map holder objects, the annotation tool will have a button to create one automatically.  I suggest you use this feature
- You will need a collider to create annotation waypoints that you can convert to lanes/lines etc.  I see in your images that you are creating in an empty space in the scene, this will not work.  The editor camera raycasts to the collider and makes a yellow indicator object so you can see where the waypoint will be made.
- I suggest you download CubeTown environment and look at the annotation there.  It has stop sign and traffic signal intersection examples that are much more simple than SF map.  Here you can practice making lanes and see how they are organized in the holder objects.
- Please let us know if you have any more issues.

wangludong, 2020-07-07:
@EricBoiseLGSVL 

I can make annotations now with your recommends.
I'll contact you if any issue comes up.Thanks a lot.

closed 2020-07-07 by wangludong