lgsvl/simulator #842: Unable to drive with autoware lgsvl in autonomous maps using vector maps

tekrajchhetri, 2020-07-06:
Hi,

I am trying to drive autonomous mode using vector maps and lgsvl simulator following the instructions from https://www.lgsvlsimulator.com/docs/autoware-instructions/#driving-by-following-vector-map. I drove a little with keyboard but still I cannot see vehicle moving and driving autonomously.

Where I want to drive, path.
[image]

I see vechile_cmd acceleration to zero.
[image]

Any help would be helpful.

Thanks.

EricBoiseLGSVL, 2020-07-06:
@hadiTab @luqiang21 Any ideas?

tekrajchhetri, 2020-07-07:
@hadiTab I was able to drive and control the vehicle using autoware. Any one wishing to achieve the same should launch the launch files provided by lgsv for simulation and launch the nodes shown in the video [https://www.youtube.com/watch?v=UaVHyV3pPSM&feature=youtu.be](https://www.youtube.com/watch?v=UaVHyV3pPSM&feature=youtu.be).

One thing to make sure is that you should be able to see blue lines and waypoints generated automatically, no need to load any waypoints.

Hope this helps!

closed 2020-07-07 by tekrajchhetri