lgsvl/simulator #844: LGSVL Simulator suddenly shut down

eungh1, 2020-07-07:
Hello, I'm using LGSVL simulator-linux64-2020.05.
I build a custom map using Unity 2019.3.3f.
So I can use a custom map into LGSVL simulator.

But, run the simulator with a custom map and simulator shut down after a few seconds.
Sometimes it shut down after a few seconds, sometimes it shut down after a few minutes.
If I use maps provided by LGSVL, the simulator does not shut down.

Is there any information needed to build a map with Unity?
When creating a custom map with Unity, I made it with reference to CubeTown.

Thanks for your great job.

EricBoiseLGSVL, 2020-07-07:
@eungh1 Thanks and welcome to lgsvl sim.  Please answer a few questions to narrow down the issue.
- What is your GPU?
- Please post nvidia-smi to see the GPU usage with your custom map
- What is the size of the map? Size of your textures?
- Please post your editor or player log, depending on the way you are running.  This has error logs so we can see if it is a code issue or resources issue.
- What is the fps you get before a crash?
- What sensors are you using?
- If you can, any screen shots of the map will help see if the map is the issue or something else.

eungh1, 2020-07-08:
@EricBoiseLGSVL Hello, I answer your questions.

- What is your GPU?
My GPU is GeForce RTX 2080.

- Please post nvidia-smi to see the GPU usage with your custom map
[image]

- What is the size of the map? Size of your textures?
Size of custom map is 1350*1150

- Please post your editor or player log, depending on the way you are running. This has error logs so we can see if it is a code issue or resources issue.
>
[Player.log](https://github.com/lgsvl/simulator/files/4890159/Player.log)

>
[Editor.log](https://github.com/lgsvl/simulator/files/4890186/Editor.log)

- What is the fps you get before a crash?
I don't know how to check the fps.

- What sensors are you using?
I'm using two cameras, four LiDARs, and gps.

- If you can, any screen shots of the map will help see if the map is the issue or something else.
[image]
[image]

EricBoiseLGSVL, 2020-07-08:
@eungh1 Ok, wow. Thanks for all the info.  This helps immensely.
- This map and setup looks great.  I do not see any issues with this.
- What shader are you using for the lines?  Is this HDRP lit?
- Do you have two 2080's?  If so, this might be an issue.  Can you force the sim exe to just use one through Nvidia control panel?
- Two cameras and 4 lidars could also be an issue.  Can you try 2 lidars and see if you have the same crash?  What is the count 16?
- I think Signal 11 is segmentation fault. This means the program is trying to access memory it doesn't have permission for.  This could be many different things. I see a bug in Unity for CubeMap api's on Vulkan.  Are your cameras using CubeMap?
- FPS can be captured with Nvidia overlays, not an issue I think
- Again, this looks fine.  I'll ask a few other engineers on this.

@rongguodong @luqiang21 @hadiTab Any of you see what could be the issue or another way to log what's going on with the Linux side?

hadiTab, 2020-07-08:
@eungh1 You may be running out of memory on your GPU. Can you try installing [nvtop](https://github.com/Syllo/nvtop) and monitoring the memory usage? i.e. start nvtop, then start the simulator and wait for it to crash while monitoring the plots for memory usage.

rongguodong, 2020-07-08:
If you use less sensors, say only 1 camera and 1 lidar, will it still shut down?

When you say "shut down", do you mean crash? Do you see any stacktrace in log files?

Is it possible to share the problematic map so that we can reproduce the crash locally?

eungh1, 2020-07-09:
@EricBoiseLGSVL @hadiTab @rongguodong 
Thank you for answering.
I don't think the number of sensors is a problem. 
When using a CubeTown or Shalun with 4 lidars and 2 cameras, simulator doesn't shut down.
When only using a custom map, simulator shut down.

@rongguodong What I mean "shut down" is "simulator window is off". My computer is still working. Only simulator shut down.
How could I share you a custom map? Would you give me your email address? It is a problem to post it in the public place.

I attached nvtop results.
>Running simulator : It was a similar compared to Shalun.
[image]

rongguodong, 2020-07-09:
My email is guodong.rong@lge.com. If it is too big to be send as attachment, you can put it somewhere online and send me the link by email.

When it crashes, please find the log file and post it here. It may be helpful. You can also post the log for a good map for comparison.

EricBoiseLGSVL, 2020-07-14:
@eungh1 I have looked at your map source and there are a few issues
- You are using a ply file. This is not supported. You need to convert the map to an fbx. See [here](https://docs.unity3d.com/Manual/3D-formats.html)
- You also needs to follow our documentation that Models folder has the mesh (.fbx's) while the child folder /Materials has the materials and textures
- Prefab folder just holds the Unity prefabs
- Scene file is good where it is
See if you fix these issues and try running the sim again

eungh1, 2020-07-15:
@EricBoiseLGSVL 
Hello, before doing what you said, I made a map by adding just a plane without a ply file.
And I sorted the folders as you said. But, the issue was not solved though.
Simulator need fbx files even if I create a map by adding planes only?

I attach the log file when there is a crash or not.
>Player log files
[Player_crash.log](https://github.com/lgsvl/simulator/files/4922890/Player_crash.log)
[Player_not_crash.log](https://github.com/lgsvl/simulator/files/4922891/Player_not_crash.log)

EricBoiseLGSVL, 2020-07-15:
@eungh1 from unity's documentation .ply files are not supported.  I would remove and replace with .fbx or .obj file.  After that we can see what the next issue could be.  We are not sure how you got that far from your screen shots because we couldn't even load your scene files.  You can even just use a unity plane and remove the files but keep the annotation.

eungh1, 2020-07-15:
@EricBoiseLGSVL 
I change the shader and issue solved it.
Thank you very much.

EricBoiseLGSVL, 2020-07-15:
@eungh1 Glad to help and happy you found the issue.

closed 2020-07-15 by EricBoiseLGSVL