lgsvl/simulator #845: Lidar points seem to have a time delay

gyp199009, 2020-07-08:
Hello,
I found that point cloud of NPC cars seem to have a time delay when they are moving ( shown as below), my simulator version is 2020.05, is there any way to solve it?

[image]

hadiTab, 2020-07-08:
There will always be some delay because the data is published after one full revolution of the lidar. The ego vehicle and dynamic objects will have moved between the time the distance is measured and the message is actually published. This delay will always be there to some extent for revolving lidars and it is not because of the simulator (real lidars have a similar delay). That being said, since our lidar runs on GPU there will be some latency for moving data between the GPU and CPU which will also add to the delay.

I'm not sure if it is possible to reduce the lag any further. @rongguodong do you have any thoughts on this?

rongguodong, 2020-07-08:
Yes, we can only make the delay smaller by improving frame rate, but will not be able to totally remove it (unless we change how we do physics update).

The way we simulate Lidar is: we render the scene many times, and read back the rendering results. We then process these results to generate point cloud and publish it out. While we are processing the rendering results, the physics update continues and thus cars continues to move. The more physics updates we have, the larger lag we will have.

gyp199009, 2020-07-10:
Do you suggest upgrading the hardware  to improve frame rate by any chance ?

martins-mozeiko, 2020-07-10:
Be sure to understand that simulator visualization is showing points captured from previous frames - to reflect same data that is sent over to ROS/Cyber. On the actual ROS messages this delay is smaller.

Basically the dataflow is following:

1) Simulate lidar on GPU and collect distances to objects
2) Transfer data from GPU to CPU
3) Process data on CPU to produce point cloud
4) Send point cloud to ROS and send point cloud back to GPU to visualize
5) Visualize point cloud on GPU - this is what you see in your screenshot.

Step 5 introduces extra delay that does not exist on ROS/Cyber messages. Only in visualization.