lgsvl/simulator #847: Simulator - choose a GPU and change ego vehicle default spawn position

dr563105, 2020-07-08:
Hello again,

I'd like to make the simulator choose a particular GPU. Is it possible to do that? If so, how? Something like in tensorflow we can choose the GPU in which the NN can be trained. 

And, is it possible to change the default spawn position of the ego vehicle in SF map, to a different location? If so, how?

Thank you.

hadiTab, 2020-07-08:
@dr563105 according to forum posts (but not officially documented) you can pass the gpu number as a commandline argument. i.e. `<path_to_binary>/simulator -gpu 0`. I only have one GPU on my machine so I haven't tested it.

Spawn points are defined in gameobjects called Spawninfo for each scene. You can open the scene in the Unity editor and change the coordinates of the gameobject. If you want to use the map in the binary build of the simulator you will need to build the map, and in the simulator webui replace the url of the map with the path to the locally built map.

EricBoiseLGSVL, 2020-07-08:
@dr563105 you can use the python api to choose a position to spawn or change the SF map spawnInfo.  We are working on tools in the web ui to let users place the ego in a particular position but it is not ready yet.

dr563105, 2020-07-10:
> @dr563105 according to forum posts (but not officially documented) you can pass the gpu number as a commandline argument. i.e. `<path_to_binary>/simulator -gpu 0`. I only have one GPU on my machine so I haven't tested it.

Unfortunately this command doesn't work. The simulator launches but with errors - `unknown argument -gpu, skipping it` and `Unknown argument 1, skipping it`. Already tried `--gpu` with either 0 or 1. By default the simulator launches in the 1st gpu we have.

closed 2020-08-06 by dr563105