lgsvl/simulator #850: Latest version source error

arslansarp, 2020-07-09:
Hi, I am trying to use LGSVL simulator from source but I couldn't manage to open it in Unity. I tried to install and uninstall several HDRP versions from package manager but it didn't work. At most I succeeded to reduce errors like 13 or so .

[image]

 What should I do ?

EricBoiseLGSVL, 2020-07-09:
@arslansarp Hello and welcome to LGSVL Simulator
- Did you install and run git lfs before you cloned?
- Did you clone not download the source zip?
- We have edits to HDRP package and it is included in the source, you do not need to update or install.
- If you have installed a different version of HDRP I suggest you delete everything and re clone with above info
- Be sure to clone maps and vehicles we have provided also.  See documentation.  They need to be in the external folder
- Once you open the project and Unity creates the Library folder, close Unity and reopen project.  This helps with some Unity issues.

arslansarp, 2020-07-10:
@EricBoiseLGSVL Thank you, it worked. My problem is solved but I have a another question. When I open the stats of unity, 

saved by batching : -5000 or so . I googled it out so people said that its not normal to  have negative values. İs it normal in lgsvl ?

EricBoiseLGSVL, 2020-07-10:
@arslansarp Glad it worked.  Yes, HDRP is not compatible with the Stats feature in the editor.  You need to download the batching code for Unity HDRP [here](https://github.com/Unity-Technologies/SRPBatcherBenchmark/blob/master/Assets/Scripts/SRPBatcherProfiler.cs)

closed 2020-07-23 by EricBoiseLGSVL