lgsvl/simulator #851: VR Cameras with LGSVL

ElliotDickman, 2020-07-10:
I'm looking to use LGSVL with a first-person VR camera. Is there an existing workflow to implement something like this?

rongguodong, 2020-07-10:
What do you mean by "VR camera"? Do you mean using VR headset like HTC Vive with our simulator? Or implement a 360 camera sensor attached to ego car?

ElliotDickman, 2020-07-10:
Using an HTC Vive with the simulator. Haven't found any resources regarding how to implement that so far.

EricBoiseLGSVL, 2020-07-10:
@ElliotDickman We don't have any documentation outlining this, you will need to look at Unity documentation, like [this](https://docs.unity3d.com/Packages/com.unity.xr.interaction.toolkit@0.9/manual/index.html), to see how to implement.  As for the Simulator, make a VR camera rig sensor and add to the vehicle sensor parameters.

closed 2020-07-31 by lemketron