lgsvl/simulator #853: Unable to install lgsvl_msgs for radar

dr563105, 2020-07-13:
Hello!

I'd like to use radar with 2020.05 simulator(https://github.com/lgsvl/simulator/blob/release-2020.05/Assets/Scripts/Bridge/Ros/Messages/Lgsvl.cs#L200). In order to use it, I need to import lgsvl_msgs(https://github.com/lgsvl/lgsvl_msgs). 

I tried installing through ROS build farm - `sudo apt install ros-$ROS_DISTRO-lgsvl_msgs` (ros2 dashing), I get an error `E: Unable to locate package ros-dashing-lgsvl-msgs` though I have repositories updated from [here](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/#setup-sources). 

Using the build from source method, I'm getting `module not found errors`. After solving a couple, I got `ModuleNotFoundError: No module named 'lark'`. I tried installing `lark-parser` but still the error exists. 

How could I solve this issue? Thank you.

dr563105, 2020-07-13:
Solved it. 

After cloning the master branch, I checked out `dashing-devel` branch. Then did `colcon build --symlink`. The package compiled correctly.

dr563105, 2020-07-13:
reopening the issue as I would like to install the package universally using the `sudo apt install ros-$ROS_DISTRO-lgsvl_msgs` command.

dr563105, 2020-07-14:
@hadiTab can you help on this as well please?

hadiTab, 2020-07-14:
You will need to wait for the next Ros dashing sync which is in a few days for it to become available through the Ros repositories.

dr563105, 2020-07-14:
ok. Can you inform me when it is done, please?

dr563105, 2020-08-31:
I think I already received the packages when I updated recently. If so, this can be closed.

closed 2020-09-01 by hadiTab