lgsvl/simulator #855: Point of spawn of NPCs and pedestrians

JockTokk, 2020-07-14:
What is start point for spawn? How can I make several point of spawn in different part of SanFrancisco?

rongguodong, 2020-07-14:
Spawn points are game objects inside the scene. It has position/rotation hard-coded there.
[image]

JockTokk, 2020-07-14:
@rongguodong sorry, I forgot to clarify I'm interested in point of spawn of NPCs and pedestrians. As I know Spawninfo is point of spawn of ego.

rongguodong, 2020-07-14:
Not only the ego. Both NPC and pedestrian use this SpawnInfo to get their spawn positions. You can refer to our examples on how to create NPC and pedestrians:
https://github.com/lgsvl/PythonAPI/blob/master/quickstart/12-create-npc-on-lane.py
https://github.com/lgsvl/PythonAPI/blob/master/quickstart/14-create-pedestrians.py

lemketron, 2020-07-14:
The Spawninfo is for the ego vehicle as described here in the [New Environment Video Tutorial for LGSVL Simulator](https://www.youtube.com/watch?v=S0w2ahhEPbE&feature=youtu.be&t=303).

The quickstart scripts get the ego spawninfo in order to position NPCs close to the ego vehicle spawn location. Otherwise NPCs will span randomly, or can be directed to spawn at any specific location using the PythonAPI (as shown in the quickstart scripts @rongguodong mentioned above.

So to make several spawn points you need to do that using the Python API.  Otherwise, NPCs will be positioned randomly in the scene if you check the setting in the UI to request random NPC vehicles and/or pedestrians.

eniseiskaya, 2020-07-14:
@rongguodong Hello, if I need to create a script that will spawn pedestrians and NPCs in a specific place on the map and time. Should I use the PythonAPI?

lemketron, 2020-07-14:
> if I need to create a script that will spawn pedestrians and NPCs in a specific place on the map and time. Should I use the PythonAPI?

@eniseiskaya Yes, as noted earlier:
> You can refer to our examples on how to create NPC and pedestrians:
> https://github.com/lgsvl/PythonAPI/blob/master/quickstart/12-create-npc-on-lane.py
> https://github.com/lgsvl/PythonAPI/blob/master/quickstart/14-create-pedestrians.py

This example will show you how to set the time of day as well:
https://github.com/lgsvl/PythonAPI/blob/master/quickstart/19-time-of-day.py

Please refer to the PythongAPI docs for additional information:
https://www.lgsvlsimulator.com/docs/python-api/

JockTokk, 2020-07-15:
@rongguodong, @lemketron thank you very much for help and for quickly responses.

closed 2020-07-15 by JockTokk