lgsvl/simulator #856: Speed signal of the ego vehicle

ashwinsarvesh, 2020-07-14:
Hello, 

I would like to get the speed of the ego vehicle from the simulation and send it to a camera that has a GPS receiver (2.5mm jack). Is this possible to get the speed of the simulated vehicle? 

The reason why I need this: This camera has ADAS functions like lane departure and forward collision warning. For these features to work, I need this speed of the simulated ego vehicle. I place this camera in front of the simulator.

EricBoiseLGSVL, 2020-07-15:
@ashwinsarvesh You can enable the sensor visualizations and have any value you need.  If you need a different UI implementation, you will need to edit simulator code or make a sensor that displays UI data.  It could even be as simple as an OnGUI().

[image]

lemketron, 2020-07-15:
You could also monitor (on ROS, ROS2, or CyberRT) the CAN Bus messages (including speed).

For example, in /apollo/canbus/chassis you will see the speed in meters per second labeled as "speed_mps". You could monitor this data programmatically and then determine how to present it to your ADAS camera.

You mentioned a GPS antenna, but generating fake GPS radio signals to trick a stationary ADAS camera into thinking it is moving is non-trivial, and not something the simulator will be able to help you with. If you can figure out how to inject the speed into the ADAS camera (or whatever it is connected to) that would be much easier.

ashwinsarvesh, 2020-07-21:
Hello,

Thank you for your reply.
I need to extract the following signals of the ego vehicle from the Simulator/Apollo to connect it to the ADAS camera.
- The vehicle speed,
- The vehicle ignition signal (12/24V),
- The Vehicle CAN bus (High/Low),
- The vehicle left turn signal and right turn signal.
Is it possible to extract this information from the ego vehicle to give it to the camera?

The attached picture contains the wiring of the camera.
[image]

rongguodong, 2020-07-21:
On the real car, these signals will be passed to your camera driver via those wires.
When you use our simulator, you may implement a bridge plugin (please refer to our ROS/ROS2/Cyber bridges). So the bridge will receive all messages from our simulator, and send the useful ones to your camera driver.

ashwinsarvesh, 2020-07-21:
Thank you for your reply
Are these data which I listed above given by the simulator or Apollo?

Regarding your idea, Unfortunately, this camera does not have a ROS driver. :/  So I guess this idea will not work. 
What do you say?

rongguodong, 2020-07-21:
I believe we have all those information available in the simulator (may need more information on that ignition signal -- what does it mean? Is it just an on/off status?)
Even if the camera dose not have a ROS driver, it must have some driver, right? You can build a bridge plugin working with that driver.

ashwinsarvesh, 2020-07-21:
ok, I will check on the driver part.
Ignition is a (12/24V) signal. When the vehicle ignition switch is turned to ACC (Accessory) position, this 12/24V is passed to this signal which powers on the camera.

ashwinsarvesh, 2020-07-22:
> I believe we have all those information available in the simulator (may need more information on that ignition signal -- what does it mean? Is it just an on/off status?)
> Even if the camera dose not [...]

How do I access this data from the simulator?

rongguodong, 2020-07-22:
Please read the code to understand how the current sensors access data and send out messages.

ashwinsarvesh, 2020-07-24:
Yes, I will do that.
But can you tell me whether I have to read lgsvl code or Apollo code to know how to access data and send out messages?
If possible, a specific part to read.

EricBoiseLGSVL, 2020-07-24:
Look at sensors and bridge code to see how we send data.  They are in their respective folders in Assets/Scripts/Sensors or /Bridge

ashwinsarvesh, 2020-07-24:
ok got it. Thanks
CanBusData is converted to Apollo specific Apollo.ChassisMsg type
Is it being published as a ROS message, it should be, right? Otherwise, how can it be sent over the ROS bridge.
Am I right?

ashwinsarvesh, 2020-07-27:
Hello,

I am planning to use [socketcan_bridge](http://wiki.ros.org/socketcan_bridge?distro=noetic) to send CAN messages received on the ROS topic /apollo/canbus/chassis to the SocketCAN device which is a CAN hardware connected to a PC.

This socket_bridge converts can messages received on a ROS topic to can messages and sends out to hardware.
What I want to know is, in the topic /apollo/canbus/chassis are can the messages received as ROS topic?

EricBoiseLGSVL, 2020-07-28:
@ashwinsarvesh We don't support Socketcan bridge currently but I think you can create a custom bridge to subscribe to the messages.  Right @hadiTab ?

ashwinsarvesh, 2020-07-28:
@EricBoiseLGSVL Can you elaborate what you mean by not support Socketcan_bridge?
I thought I could install socketcan_bridge and then use it to send the "can_msgs/Frame" messages from the simulator (after converting from CanBusData type). I mean, I thought both were independent and Socketcan_bridge doesn't have to do anything with the simulator.

And by a custom bridge, you mean custom bridge plugin?

EricBoiseLGSVL, 2020-07-28:
@ashwinsarvesh Yes we support ROS, ROS2 and CyberRT.  This new bridge type will need to be added or create a custom solution.

ashwinsarvesh, 2020-07-28:
@EricBoiseLGSVL I get it now. ROS bridges and Cyber RT bridges are adapted to be supported by LGSVL but socketcan_bridge is not adapted. Is my understanding correct?

EricBoiseLGSVL, 2020-07-28:
@hadiTab is the expert on this but yes you are correct

ashwinsarvesh, 2020-07-28:
Can you tell me what I have to refer to implement a custom bridge plugin?

EricBoiseLGSVL, 2020-07-28:
Sorry, this feature has not released yet but it should be soon.  In the meantime you can look to how we implemented the other bridge types and start to create your own in source.

ashwinsarvesh, 2020-07-28:
> @ashwinsarvesh <https://github.com/ashwinsarvesh> from what I can tell
> from what you have described you don't need a new bridge plugin, you need
> to add a new ROS1 message of type can_msgs/Frame. You would then run the
> rosbridge and socketcan_bridge on the same machine. rosbridge will publish
> the can_msgs/Frame message which I believe socketcan_bridge will
> subscribe to and will then pass along to your device.
>
> To publish the new message you need to:
>
>    - Add the can_msgs/Frame to the list of message types supported here
>    This defines the ros side of the message.
>    - Add a conversion function to convert CanBusData
>    to can_msgs/Frame type message. See similar conversion functions here
>    - Add a condition here
>    for the writer function to use the conversion function when using ROS1.
>
> Now make sure you add the CanBus sensor to your vehicle json config, and
> that you select the ROS1 bridge. You should now recieve the can_msgs/Frame
> message on the topic defined in the json. If you run the rosbridge on the
> machine connected to your CAN device you will be publishing the
> can_msgs/Frame message on that machine and the socketcan_bridge should be
> able to subscribe to it and communicate with your CAN device.

Thank you for your reply
I'll surely check this method
In this method,looks like socketcan_bridge can work independently without
the need of LGSVL support. Am I right?

ashwinsarvesh, 2020-07-29:
Thank you for your reply.
I'll check this method
Where can I find the below details for the individual signals(for eg. Speed, left turn signal, right turn signal)
As the "can_msgs/Frame" asks for the below details. It would be helpful if I can find these.
[image]

ashwinsarvesh, 2020-07-31:
Hello,

I am sorry if I was not clear with my previous question.
For each can signal produced by the simulator(eg. speed, left and right turn signal), 

- where can I find It's message information like CAN identifiers and so on. 
- Is the data sent out in bytes?

I need this info to convert to the can_msgs/Frame type. As of now, I can just see the can bus data [here](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Data/CanBusData.cs)