lgsvl/simulator #857: Modify the CubeTown environment but couldn't play the scene

duongtran11, 2020-07-15:
Hi, I just cloned the CubeTown environment into Unity as described in the Build Instructions (https://www.lgsvlsimulator.com/docs/build-instructions/) to add some 3d objects. 

[image]

When I hit the play button the game display appears like this
[image]
I couldn't rotate and pan the game scene, the car didn't appears
When I build the simulator, the environment_CubeTown file has 0Kb data as show below
[image]
My goal is to modify the CubeTown environment then build a simulator from it.
Any help would be appreciated!

EricBoiseLGSVL, 2020-07-15:
@duongtran11 Did you clone a vehicle?  You will need a vehicle to run a simulation

duongtran11, 2020-07-16:
Hi Eric, I did clone a vehicle

[image]

I just found out the problem with the environment_CubeTown has 0kb data is because I didn't install Linux Support Build (I'm running Unity on Windows so I thought Windows support build would be enough). Now when I hit play the scene it came out like this

[image]

Still couldn't see the car and as you can see the road is disappeared

EricBoiseLGSVL, 2020-07-16:
@duongtran11  Again you need to run the simulator in Editor from Loader.scene.  We are working on developer settings to run a sim from the environment scene you have open but that will not release for a bit.

duongtran11, 2020-07-17:
I think I misunderstood something here. 
- Do I need to build the simulator first (by Simulator -> Build) in order to run it from LoaderScene. Because when you open LoaderScene and hit the play button, you got to open browser and then add map and vehicles meaning that the map need to be build in your local
- In this video https://www.youtube.com/watch?v=S0w2ahhEPbE (at 5:17) the simulator runs right from the scene with out LoaderScene

EricBoiseLGSVL, 2020-07-17:
Yes, in 2020.05 development settings is in flux.  Simulator -> Development Settings.  You needs a built vehicle to run it.  If you don't have a asset bundle for the vehicle selected it will not work.  We are working on development settings updates and will release soon

duongtran11, 2020-07-21:
I made it, thank you!

closed 2020-07-21 by duongtran11