lgsvl/simulator #859: Use the lgsvlsimulator to open the browser,Web page flashing

GuoliangPeng, 2020-07-16:
My computer:
windows10
[image]
I downloaded the binary lgsvlsimulator from github, on windows platform.Then I click to “Open Browser..."，This is the problem：
[image]

GuoliangPeng, 2020-07-16:
Web page reloads and flashes

dr563105, 2020-07-16:
It is recommended to use minimum 8GB GPU. However, the issue can be other things, can you provide us the player log? location can be found in the faq.

EricBoiseLGSVL, 2020-07-16:
I would also use Ctrl+Shift+R to clear cache.  Do you have an extensions/ad blocks running?

EricBoiseLGSVL, 2020-07-16:
@zelenkovsky maybe David would know?

EricBoiseLGSVL, 2020-07-16:
It looks like there is a redirect loop happening in the browser. This can happen when there is a mismatch in the redirect setup and the redirect information in the response. The server redirects the request with a message that says to redirect back to the same site. Thus, this causes a never ending loop.
 
Are you trying to go to localhost with a port number?

GuoliangPeng, 2020-07-17:
> It is recommended to use minimum 8GB GPU. However, the issue can be other things, can you provide us the player log? location can be found in the faq.
Mono path[0] = 'E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/simulator_Data/Managed'
Mono config path = 'E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/MonoBleedingEdge/etc'
Loading player data from E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/simulator_Data/data.unity3d
Initialize engine version: 2019.3.3f1 (7ceaae5f7503)
[Subsystems] Discovering subsystems at path E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: NVIDIA GeForce GTX 1660 (ID=0x2184)
    Vendor:   
    VRAM:     5991 MB
    Driver:   27.21.14.5167
Begin MonoManager ReloadAssembly
- Completed reload, in  0.065 seconds
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
D3D11 device created for Microsoft Media Foundation video decoding.
<RI> Initializing input.

New input system (experimental) initialized
<RI> Input initialized.

<RI> Initialized touch support.

WARNING: Shader Unsupported: 'Shader Graphs/CloudGenerator' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
WARNING: Shader Unsupported: 'Shader Graphs/VehicleLightsShader' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
UnloadTime: 23.806400 ms
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
found manifest at Controllables\controllable_ChargingStation\manifest 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/simulator_Data/Mono/data-00000176C9C85BB0.dll
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
found manifest at Controllables\controllable_TrafficCone\manifest 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/simulator_Data/Mono/data-000001769E6D2960.dll
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
found manifest at Sensors\sensor_CheckControlSensor\manifest 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/simulator_Data/Mono/data-00000176C9E35C80.dll
found manifest at Sensors\sensor_VelodyneLidarSensor\manifest 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library E:/lgsvlsimulator-windows64-2020.05/lgsvlsimulator-windows64-2020.05/simulator_Data/Mono/data-00000176CA12D010.dll
loaded 3 NPCs behaviours and 6 NPC models in 00:00:00.2085132 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

NullReferenceException: Object reference not set to an instance of an object
  at System.Net.AutoWebProxyScriptEngine.InitializeRegistryGlobalProxy () [0x0005b] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.AutoWebProxyScriptEngine.GetWebProxyData () [0x00007] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.WebProxy.UnsafeUpdateFromRegistry () [0x0001a] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.WebProxy..ctor (System.Boolean enableAutoproxy) [0x0000d] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.WebProxy.CreateDefaultProxy () [0x00012] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.Configuration.DefaultProxySectionInternal.GetSystemWebProxy () [0x00000] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.Configuration.DefaultProxySectionInternal.GetDefaultProxy_UsingOldMonoCode () [0x00036] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.Configuration.DefaultProxySectionInternal.GetSection () [0x00015] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.WebRequest.get_InternalDefaultWebProxy () [0x00022] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.HttpWebRequest..ctor (System.Uri uri) [0x0008d] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at (wrapper remoting-invoke-with-check) System.Net.HttpWebRequest..ctor(System.Uri)
  at System.Net.HttpRequestCreator.Create (System.Uri uri) [0x00000] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
[... 85 lines cut ...]
System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1:Start(<Dispatch>d__5&)
Nancy.Routing.DefaultRequestDispatcher:Dispatch(NancyContext, CancellationToken)
Nancy.<InvokeRequestLifeCycle>d__22:MoveNext()
System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1:Start(<InvokeRequestLifeCycle>d__22&)
[... cut ...]
System.Threading.Tasks.Task:ExecutionContextCallback(Object)
System.Threading.ExecutionContext:RunInternal(ExecutionContext, ContextCallback, Object, Boolean)
System.Threading.ExecutionContext:Run(ExecutionContext, ContextCallback, Object, Boolean)
System.Threading.Tasks.Task:ExecuteWithThreadLocal(Task&)
System.Threading.Tasks.Task:ExecuteEntry(Boolean)
System.Threading.Tasks.Task:System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
System.Threading.ThreadPoolWorkQueue:Dispatch()
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()

GuoliangPeng, 2020-07-17:
> 我还将使用Ctrl + Shift + R清除缓存。您是否正在运行扩展程序/广告块？

I cleared the cache using shortcut keys in chrome, and this problem still occurs. I downloaded Firefox and set it as the default, which also happened.

GuoliangPeng, 2020-07-17:
> 看来浏览器中正在发生重定向循环。当重定向设置和响应中的重定向信息不匹配时，可能会发生这种情况。服务器使用一条消息重定向到该请求，该消息显示为重定向回同一站点。因此，这导致永无止境的循环。
> 
> 您是否要使用端口号进入localhost？

It may seem to be this. I input ”http://localhost:8080/“ in the browser to open, this problem still occurs

EricBoiseLGSVL, 2020-07-17:
it should be localhost:8080

GuoliangPeng, 2020-07-17:
> it should be localhost:8080

Yes. I wrote it wrong in the front, using yours, this problem still occurs

GuoliangPeng, 2020-07-17:
It looks like the webpage keeps loading in a loop. Even if I clear the browser data, uninstall and reinstall, the problem is about to occur

[image]

EricBoiseLGSVL, 2020-07-17:
@zelenkovsky Can you add David here so he can look at this issue?

thisisdavidfrank, 2020-07-17:
> It looks like the webpage keeps loading in a loop. Even if I clear the browser data, uninstall and reinstall, the problem is about to occur
> 

Would you be able to provide the Network logs (from your browser's developer tools)? It appears that the token being send in the url is being rejected by the server, with another token....which is being rejected again, and so on and so forth. Maybe the Network logs have some info on what the error response is.

GuoliangPeng, 2020-07-20:
> > It looks like the webpage keeps loading in a loop. Even if I clear the browser data, uninstall and reinstall, the problem is about to occur
> 
> Would you be able to provide the Network logs (from your browser's developer tools)? It appears that the token being sent in the url is being rejected by the server and the server responds with another token....which is being rejected again, and so on and so forth. Maybe the Network logs have some info on what the error response is.

Is it this one? I do not use web logs
[image]
[image]

thisisdavidfrank, 2020-07-20:
> Is it this one? I do not use web logs

@GuoliangPeng, yes, that is the one. So, we need to get the network logs when the weird 'looping' starts to happen. So, when you launch your browser from the Simulator, please open your Developer Tools. This will start recording the network activity. After about 10 seconds (that is, after getting a substantial number of redirect loops), please click on the "Export HAR..." icon in the Network tab (as shown in the image below). This will let you save the network logs as a file, and share it here. Thank you.

[image]

GuoliangPeng, 2020-07-20:
> > 是这个吗？我不使用网络日志
> 
> @GuoliangPeng，是的。因此，我们需要在怪异的“循环”开始发生时获取网络日志。因此，当您从模拟器启动浏览器时，请打开您的开发人员工具。这将开始记录网络活动。大约10秒钟后（即，在获得大量重定向循环之后），请单击“网络”选项卡中的“导出HAR ...”图标（如下图所示）。这将使您将网络日志另存为文件，并在此处共享。谢谢。
> 
> [image]

Maybe this：
[localhost .txt](https://github.com/lgsvl/simulator/files/4945685/localhost.txt)

GuoliangPeng, 2020-07-20:
> @GuoliangPeng, yes, that is the one. So, we need to get the network logs when the weird 'looping' starts to happen. So, when you launch your browser from the Simulator, please open your Developer [...]
Maybe this
[localhost.log](https://github.com/lgsvl/simulator/files/4945708/localhost.log)
[localhost .txt](https://github.com/lgsvl/simulator/files/4945709/localhost.txt)

GuoliangPeng, 2020-07-20:
> > 是这个吗？我不使用网络日志
> 
> @GuoliangPeng，是的。因此，我们需要在怪异的“循环”开始发生时获取网络日志。因此，当您从模拟器启动浏览器时，请打开您的开发人员工具。这将开始记录网络活动。大约10秒钟后（即，在获得大量重定向循环之后），请单击“网络”选项卡中的“导出HAR ...”图标（如下图所示）。这将使您将网络日志另存为文件，并在此处共享。谢谢。
> 
> [image]

Maybe this（The contents of these two files are the same）：
[localhost .txt](https://github.com/lgsvl/simulator/files/4945714/localhost.txt)
[localhost.log](https://github.com/lgsvl/simulator/files/4945716/localhost.log)

thisisdavidfrank, 2020-07-20:
@GuoliangPeng, yes both files are the same :), but they are both correct.

It appears that the browser is doing a full refresh and thus losing the previous session's logs in the Network tab, and thus we don't know what exactly is causing this redirect to happen. Would you be able to do the same log capture, but this time with both the 'Preserve log' and 'Disable cache' checkboxes selected, please (as shown in image  below)?

The 'Preserve log' checkbox will help capture logs across all sessions of the browser regardless of how many times the browser refreshes. And the 'Disable cache' checkbox will help bypass the browser cache and help us understand what responses are being sent by the server.

[image]

GuoliangPeng, 2020-07-21:
> @GuoliangPeng，是的，两个文件都是相同的:)，但它们都是正确的。
> 
> 看来浏览器正在执行完全刷新，因此丢失了“网络”选项卡中上一个会话的日志，因此我们不知道正是导致此重定向发生的原因。您是否可以执行相同的日志捕获，但是这次请同时选中“保留日志”和“禁用缓存”复选框（如下图所示）？
> 
> “保留日志”复选框将帮助捕获浏览器所有会话中的日志，而不管浏览器刷新多少次。“禁用缓存”复选框将帮助绕过浏览器缓存，并帮助我们了解服务器正在发送哪些响应。
> 
> [image]

[account.lgsvlsimulator.com.txt](https://github.com/lgsvl/simulator/files/4950674/account.lgsvlsimulator.com.txt)
[account.lgsvlsimulator.com.log](https://github.com/lgsvl/simulator/files/4950675/account.lgsvlsimulator.com.log)

[image]

GuoliangPeng, 2020-07-21:
The problem still occurs, very distressed

thisisdavidfrank, 2020-07-21:
@GuoliangPeng, the problem seems to be an old and expired token that is causing an "internal server error" in the Simulator.  Let's try one of the following methods to fix this problem (Please do capture your network logs (with 'Preserve log' and 'Disable cache' turned on) for both these methods):

_Method 1:_
Close and re-open your Simulator.
Next, open your browser and go to https://account.lgsvlsimulator.com/.
Important: If you are signed in on this website, then you would need to logout. To logout, click on the username on the top right corner of the page and click on 'Sign out'.
After sign out, on the same browser, go to http://localhost:8080/. (NOTE: Copy-Paste this URL in the browser and hit enter. Do not click on the simulator's Open Browser button.)
Now, once you are on localhost, it should load properly without doing the weird looping.
If this doesn't, then try Method 2.

_Method 2:_
Close and re-open your Simulator.
If you are using Chrome, open an Incognito window. If you are using Firefox, open a Private window.
Now, go to http://localhost:8080/. (NOTE: Copy-Paste this URL in the browser and hit enter. Do not click on the simulator's Open Browser button.)
This should now automatically take you to https://account.lgsvlsimulator.com/.
Sign in on this website. Once you are signed in, it will automatically take you back to http://localhost:8080/.
Now, once you are on localhost, it should load properly without doing the weird looping.
If this doesn't work either, then please send me your Network logs.

GuoliangPeng, 2020-07-22:
> @GuoliangPeng, the problem seems to be an old and expired token that is causing an "internal server error" in the Simulator. Let's try one of the following methods to fix this problem (Please do [...]

I tried these two methods and the problem still occurs：
[Method1 account.lgsvlsimulator.com.log](https://github.com/lgsvl/simulator/files/4957381/Method1.account.lgsvlsimulator.com.log)
[Method2 account.lgsvlsimulator.com.log](https://github.com/lgsvl/simulator/files/4957382/Method2.account.lgsvlsimulator.com.log)

thisisdavidfrank, 2020-07-22:
Hey @GuoliangPeng, this is definitely weird because I tried to recreate the same issue by changed my language to Chinese and tried to login. But this issue didn't happen to me. I also tried to login with your credentials since I thought maybe there was something wrong with your account, but this issue didn't happen to me again.

So, there is definitely something on your machine that is causing this issue to happen.

1. Can you try cleaning your Simulator data folder. On your Windows machine, you need to go to:
**C:\Users\_your_username_here_\AppData\LocalLow**
and delete the folder that says "LG Silicon Valley Lab".

2. Do you have an older version of the Simulator? Can you download the latest version from here and use that instead of the version that you have:
https://github.com/lgsvl/simulator/releases/tag/2020.05

3. Lastly, can you confirm that your Windows machine's clock is correct. Because if your clock is in the future, then the token will expire immediately. So, please check the clock on your Windows machine and make sure it is set correctly.

4. Finally, launch your Simulator, and click on the 'Open Browser' button. Hopefully this works properly for you this time.

5. If this happens again, then maybe the last resort is to try to create a new user account on https://account.lgsvlsimulator.com/, and try with that.

GuoliangPeng, 2020-07-22:
> Hey @GuoliangPeng, this is definitely weird because I tried to recreate the same issue by changed my language to Chinese and tried to login. But this issue didn't happen to me. I also tried to login with your credentials since I thought maybe there was something wrong with your account, but this issue didn't happen to me again.
> 
> So, there is definitely something on your machine that is causing this issue to happen.
> 
> 1. Can you try cleaning your Simulator data folder. On your Windows machine, you need to go to:
>    **C:\Users_your_username_here_\AppData\LocalLow**
>    and delete the folder that says "LG Silicon Valley Lab".
> 2. Do you have an older version of the Simulator? Can you download the latest version from here and use that instead of the version that you have:
>    https://github.com/lgsvl/simulator/releases/tag/2020.05
> 3. Lastly, can you confirm that your Windows machine's clock is correct. Because if your clock is in the future, then the token will expire immediately. So, please check the clock on your Windows machine and make sure it is set correctly.
> 4. Finally, launch your Simulator, and click on the 'Open Browser' button. Hopefully this works properly for you this time.
> 5. If this happens again, then maybe the last resort is to try to create a new user account on https://account.lgsvlsimulator.com/, and try with that.
> 
> If the last step doesn't fix the issue for you, then it is possible that you have a proxy server setup. Maybe this can help you with your issue:
> [mono/mono#10030](https://github.com/mono/mono/issues/10030)
> Either ProxyEnable should be false, or if it is true, then ProxyServer string should exist and have a value.

The problem was solved just now. 
I have tried steps 1-5, but no effect.
Finally,I solved the problem with the following method(I feel lucky):
open "windows settings"  --> open "Network and Internet"  -->  choose "proxy or agent" -->Turn off automatic proxy setting and manual proxy setting.
Then, the browser interface is displayed normally. I can't wait to start using the simulator.Thank you for your help! @thisisdavidfrank 
[image]

EricBoiseLGSVL, 2020-07-23:
@GuoliangPeng Glad you found the solution and posted for other users.  Thanks @thisisdavidfrank for the help

closed 2020-07-23 by EricBoiseLGSVL

thisisdavidfrank, 2020-07-23:
@GuoliangPeng, please remember to edit your comments and remove all your network log attachments. A couple of them have your login info, and you don't want that to remain exposed :).