lgsvl/simulator #860: Enabling multiple cameras in LGSVL with Autoware

SenRamakri, 2020-07-16:
Hello!,
Is there any documentation/pointers on how to enable multiple cameras on LGSVL with Autoware.AI?
Thanks.

hadiTab, 2020-07-18:
@SenRamakri for adding multiple cameras all you need to is add multiple camera sensors to the sensor configuration in the simulator. A sample sensor configuration for AutowareAI (with only one camera) is available [here](https://www.lgsvlsimulator.com/docs/autoware-json-example/).

If you want to add more cameras just add more instances of the camera sensor in the JSON and make sure to provide the correct transform and a unique topic name and sensor name. For example:

```json
  {
    "type": "Color Camera",
    "name": "Camera_X",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "JpegQuality": 75,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 1000,
      "Topic": "/simulator/camera_x/image/compressed",
      "Frame": "camera_x"
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
```

That being said, I don't think AutowareAI supports multiple cameras. They do use Yolo and SSD for object detection using cameras but last time I checked it was not possible to pass on these detected obstacles to the planner to avoid collision. You should probably ask the AutowareAI folks to get more information on what to do on the Autoware side but I suspect you will need to implement it yourself.

SenRamakri, 2020-07-18:
@hadiTab - Thanks much for your reply and it really helped. I'll see if I can get more info on multiple camera support with Autoware.AI.

SenRamakri, 2020-07-23:
Since the question has been answered, I'll go ahead close this issue. Thanks for the quick response @EricBoiseLGSVL and @hadiTab

closed 2020-07-23 by SenRamakri