lgsvl/simulator #861: Apollo5.5 is support for Volta and Turing NOW?

yumianhuli2, 2020-07-17:
https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/
The current version of Apollo does not support Volta and Turing architectures (this includes Titan V and RTX 2080 GPUs).
So Apollo5.5 is support for Volta and Turing NOW?

EricBoiseLGSVL, 2020-07-17:
@yumianhuli2 yes. We are working on Apollo 5.5 support currently.  @rongguodong is the documentation released yet for 5.5?

rongguodong, 2020-07-17:
@yumianhuli2 Please refer to this instruction for Apollo 5.5: https://www.lgsvlsimulator.com/docs/apollo-master-instructions/

yumianhuli2, 2020-07-18:
> @yumianhuli2 yes. We are working on Apollo 5.5 support currently. @rongguodong is the documentation released yet for 5.5?

Thank you I will try！And I hope run the simulator and algorithm on one PC with 6 gigabytes of video memory：）

rongguodong, 2020-07-18:
@yumianhuli2 Note that the latest master has bugs for perception (see https://github.com/ApolloAuto/apollo/issues/11818). You may want to test "pre6" branch instead of "master" branch.

closed 2020-07-25 by yumianhuli2

yumianhuli2, 2020-07-31:
> @yumianhuli2 Note that the latest master has bugs for perception (see [ApolloAuto/apollo#11818](https://github.com/ApolloAuto/apollo/issues/11818)). You may want to test "pre6" branch instead of "master" branch.
Hello！
whether the master branch has fixed this bug or not, Github cannot clone it. There is no pre6 branch on gitee!

lemketron, 2020-07-31:
You can find the "pre6" branch here: https://github.com/ApolloAuto/apollo/tree/pre6

yumianhuli2, 2020-08-01:
> You can find the "pre6" branch here: https://github.com/ApolloAuto/apollo/tree/pre6
> Of course you should be able to clone the project and just `git checkout pre6` to switch to this branch.

can not clone so big repo:)

yumianhuli2, 2020-08-01:
> @yumianhuli2 Please refer to this instruction for Apollo 5.5: https://www.lgsvlsimulator.com/docs/apollo-master-instructions/

I can not open this page https://www.lgsvlsimulator.com/docs/apollo-master-instructions/

EricBoiseLGSVL, 2020-08-03:
@yumianhuli2 You can't open our LGSVL documentation?