lgsvl/simulator #864: LGSVL working with Apollo Master branch (July 18 2020)

ntutangyun, 2020-07-18:
Hi team,

I pulled the latest Apollo master branch and LGSVL this afternoon.

The perception no longer works anymore.

I managed to make a temporary fix for this. 

Please help check out the steps at 
https://github.com/ApolloAuto/apollo/issues/11846

and see if there is a need to update the documentation at 
https://www.lgsvlsimulator.com/docs/apollo-master-instructions/

thanks.

rongguodong, 2020-07-18:
Apollo master is not stable now. You can refer [Issue 11818](https://github.com/ApolloAuto/apollo/issues/11818) for more details.
Before master is stable, you may want to try "pre6" branch, instead of "master" branch.

ntutangyun, 2020-07-19:
@rongguodong I see. alright. thanks for the clarification.

closed 2020-07-19 by ntutangyun

rongguodong, 2020-07-20:
BTW: We are going to support modular tests to bypass the perception modules, i.e. simulator directly publish perception results to "/apollo/perception/obstacles" and "/apollo/perception/trafficlight", so that you can still test other modules even with broken perceptions. :)

ntutangyun, 2020-07-21:
> i.e. simulator directly publish perception results to "/apollo/perception/obstacles" and "/apollo/perception/trafficlight", so that you can still test other modules even with broken perceptions. :)

@rongguodong yeah. I did that previously. you just need to add two custom sensors for LGSVL and let them publish directly to those two perception channels.

ntutangyun, 2020-07-21:
> BTW: We are going to support modular tests to bypass the perception modules, i.e. simulator directly publish perception results to "/apollo/perception/obstacles" and "/apollo/perception/trafficlight", so that you can still test other modules even with broken perceptions. :)

I did this already. I added two custom sensors for this. 

PerceptionObstacleSensor.cs
---

```c#
using System;
using System.Linq;
using System.Collections.Generic;
using UnityEngine;
using Simulator.Bridge;
using Simulator.Map;
using Simulator.Utilities;
using Simulator.Sensors.UI;
using PerceptionObstacles = apollo.perception.PerceptionObstacles;
using PerceptionObstacle = apollo.perception.PerceptionObstacle;

namespace Simulator.Sensors
{
    [SensorType("Perception Obstacle Sensor", new[] { typeof(PerceptionObstacles) })]
    public class PerceptionObstacleSensor : SensorBase
    {
        struct TrackedObstacle
        {
            public double firstTrackTime;
        }

        [SensorParameter]
        [Range(1f, 100f)]
        public float Frequency = 10.0f;

        [SensorParameter]
        [Range(1f, 1000f)]
        public float MaxDistance = 200.0f;

        [SensorParameter]
        public bool IgnoreMapOrigin = false;

        public RangeTrigger rangeTrigger;

        private IBridge Bridge;
        private IWriter<PerceptionObstacles> obstacleWriter;

        private Dictionary<uint, PerceptionObstacle> Detected = new Dictionary<uint, PerceptionObstacle>();
        private Dictionary<uint, TrackedObstacle> trackedObstacles = new Dictionary<uint, TrackedObstacle>();

        public override SensorDistributionType DistributionType => SensorDistributionType.HighLoad;
        MapOrigin MapOrigin;

        private uint seqId;
        private double nextSend;

        private void Awake()
        {
            MapOrigin = MapOrigin.Find();
        }

        void Start()
        {
            rangeTrigger.SetCallbacks(WhileInRange);
            rangeTrigger.transform.localScale = MaxDistance * Vector3.one;
            nextSend = SimulatorManager.Instance.CurrentTime + 1.0f / Frequency;
        }

        void Update()
        {
            double currentTime = SimulatorManager.Instance.CurrentTime;

[... 392 lines cut ...]
                }
            }
        }

        public override void OnBridgeSetup(IBridge bridge)
        {
            Bridge = bridge;
            trafficLightWriter = Bridge.AddWriter<TrafficLightData>(trafficLightTopic);
        }

        public override void OnVisualize(Visualizer visualizer)
        {
            //
        }

        public override void OnVisualizeToggle(bool state)
        {
            //
        }
    }
}

```

Hope it helps ~