lgsvl/simulator #867: Does LGSVL support runtime control?

ForresterLu, 2020-07-20:
I am doing a project which need to runtime control the LGSVL environment, like weather, NPC vehicles, pedestrian and etc. For example, when a ECO vehicle is running, I can let the city rain by invoking an API, can I do this in runtime?

lemketron, 2020-07-20:
Yes, please check out the Python API documentation [here](https://www.lgsvlsimulator.com/docs/python-api/) and specifically the quickstart scripts on [GitHub](https://github.com/lgsvl/PythonAPI/tree/master/quickstart).

ForresterLu, 2020-07-20:
Thanks for your reply. I have already check the Python APIs, but once I set the control signal to ECO vehicle and call sim.run() to let it run, I can not call APIs to control the environment. So my question is, can I first let the ECO vehicle run, then call APIs to control the weather, NPC vehicles, pedestrian and etc.
I am not sure because I saw in Python API documentation it says "Only one client can be connected to simulator at a time." 
[image]

lemketron, 2020-07-20:
According to the [API execution diagram](https://www.lgsvlsimulator.com/docs/python-api/#simulation) you cannot make other API calls until sim.run returns (either from the time_limit specified in sim.run being exceeded or a callback like from a collision).

You'll probably want to specify a time_limit (e.g. `sim.run(time_limit = 1.0)`) to sim.run in order to have the simulation run for a second and then return so you can (if needed) control the environment, then call sim.run again.

We are looking into other options to control the simulator from the Python API while it is running without having to specify a time_limit to sim.run.

ForresterLu, 2020-07-20:
> According to the [API execution diagram](https://www.lgsvlsimulator.com/docs/python-api/#simulation) you cannot make other API calls until sim.run returns (either from the time_limit specified in [...]

It's very nice to see your reply, but what I want to do is when the sim.run(time_limit = 10) (for example), I want to call another API to control the environment at the same time. 
It cannot be controlled at the same time, I think we need to do this by connecting multiple client to simulator or multithreaded control, I am not sure.

lemketron, 2020-07-20:
> It's very nice to see your reply, but what I want to do is when the sim.run(time_limit = 10) (for example), I want to call another API to control the environment at the same time.
> It cannot be [...]

The simulator does not support multiple clients or multithreaded access at this time.

closed 2020-07-20 by ForresterLu