lgsvl/simulator #868: Connect to Simulator with IP of

ForresterLu, 2020-07-20:


ForresterLu, 2020-07-20:
Hello everyone, I try to connect to the simulator by the following command:
`sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)`
and I can sucessfully connect to the simulator, it's also fine when I change "127.0.0.1" to "localhost".

But when I try to connect with the IP of the same computer by the command, "192.168.1.55" is my computer IP:
`sim = lgsvl.Simulator(address="192.168.1.55", port=8181)` , it dose not work.

And the error message is:
`D:\Anaconda3\python.exe "D:/RTCM COPY/Autonomous Car/LGSVL/LGSVLProject/PythonAPI/quickstart/01-connecting-to-simulator.py"
Exception in thread Thread-1:
Traceback (most recent call last):
  File "D:\Anaconda3\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "d:\rtcm copy\autonomous car\lgsvl\pythonapi\lgsvl\remote.py", line 28, in run
    self.loop.run_until_complete(self.process())
  File "D:\Anaconda3\lib\asyncio\base_events.py", line 468, in run_until_complete
    return future.result()
  File "d:\rtcm copy\autonomous car\lgsvl\pythonapi\lgsvl\remote.py", line 36, in process
    self.websocket = await websockets.connect(self.endpoint, compression=None)
  File "C:\Users\LuChengjie\AppData\Roaming\Python\Python36\site-packages\websockets\py35\client.py", line 12, in __await_impl__
    transport, protocol = await self._creating_connection
  File "D:\Anaconda3\lib\asyncio\base_events.py", line 778, in create_connection
    raise exceptions[0]
  File "D:\Anaconda3\lib\asyncio\base_events.py", line 765, in create_connection
    yield from self.sock_connect(sock, address)
  File "D:\Anaconda3\lib\asyncio\selector_events.py", line 450, in sock_connect
    return (yield from fut)
  File "D:\Anaconda3\lib\asyncio\selector_events.py", line 480, in _sock_connect_cb
    raise OSError(err, 'Connect call failed %s' % (address,))
ConnectionRefusedError: [Errno 10061] Connect call failed ('192.168.1.55', 8181)
`
Dose anyone know what's the matter?

lemketron, 2020-07-20:
According to the [Python API documentation](https://www.lgsvlsimulator.com/docs/python-api/#quickstart), the Simulator by default listens for connections on port 8181 on localhost.

You can see this if you look in the config.yml file in the simulator directory: `api_hostname: "localhost"`.

If you change "localhost" to your IP address (e.g. "192.168.1.55") and re-launch the simulator you'll find that it should respond on that interface.

ForresterLu, 2020-07-20:
> According to the [Python API documentation](https://www.lgsvlsimulator.com/docs/python-api/#quickstart), the Simulator by default listens for connections on port 8181 on localhost.
> 
> You can see this if you look in the "config.yml" file in the simulator directory: `api_hostname: "localhost"`.
> 
> If you change "localhost" to your IP address (e.g. "192.168.1.55") in the "config.yml" file and re-launch the simulator you'll find that it should respond on that interface.

Thanks for your replay, I modified the api_hostname to my computer IP in config.yml file as you suggested, but the question remains the same. You can see the config.yml as followed:
`hostname: "localhost"
port: 8080
headless: false
read_only: false
api_hostname: "192.168.1.55"
api_port: 8181`

Any idea about that, please kindly let me know, thank you.

lemketron, 2020-07-20:
I tried this on Ubuntu and it worked for me after changing the "config.yml" file.

It looks like you're on Windows, so I'll defer to @EricBoiseLGSVL to see what he suggests.

ForresterLu, 2020-07-20:
Yes, I am on Windows 10, and I meet the same issue on my another Windows 10 computer.

EricBoiseLGSVL, 2020-07-20:
@ForresterLu Please check that no other application is using this port

lemketron, 2020-07-20:
[This article](https://www.howtogeek.com/howto/28609/how-can-i-tell-what-is-listening-on-a-tcpip-port-in-windows/) might be helpful in figuring out if the port is in use (using `netstat -ab` and other available tools); make sure that the simulator is the only process using (or attempting to use) it. It looks like this might also show which interface (e.g. IP address) the port is opened on.

ForresterLu, 2020-07-21:
@EricBoiseLGSVL 
> @ForresterLu Please check that no other application is using this port

@lemketron 
> [This article](https://www.howtogeek.com/howto/28609/how-can-i-tell-what-is-listening-on-a-tcpip-port-in-windows/) might be helpful in figuring out if the port is in use (using `netstat -ab` and [...]

Thanks for your reply, I check the open TCP/IP Ports in Windows using `netstat -ab` ,here is the result:
[image]
The simulator is using the port 8181 and IP 127.0.0.1 just what I set in the "config.yml" file.
And when I connect to the simulator using `sim = lgsvl.Simulator(address="127.0.0.1", port=8181)` , it worked without error.
But when I changed the IP to my computer's IP which is `192.168.1.55` using the following command 
`sim = lgsvl.Simulator(address="192.168.1.55", port=8181)` , it still did not work.

I see in the [LGSVL API document](https://www.lgsvlsimulator.com/docs/python-api/) the connection is established using **websocket** , as far as I know, the connect should be established using both `localhost`(`127.0.0.1`) and `192.168.1.55` which is my computer's IP. 
But the truth is it cannot work when I change `localhost` or `127.0.0.1` to `192.168.1.55`.

lemketron, 2020-07-21:
It looks like you updated api_hostname to "192.168.1.55" but the output of netstat isn't reflecting that. 
Did you quit and restart the simulator after making this change?

EricBoiseLGSVL, 2020-07-28:
@ForresterLu Did a restart after the change fix the issue?

ForresterLu, 2020-07-28:
> @ForresterLu Did a restart after the change fix the issue?

Hello @EricBoiseLGSVL , I just changed the api_hostname to my computer host `192.168.1.55` in `config.yml`, see the following screenshot:
[image]
then I restarted the simulator and tried to connect to the simulator by the following command:
`sim = lgsvl.Simulator(address="192.168.1.55", port=8181)`

however, it still didn't work and printed the error message as following:
[image]

lemketron, 2020-07-28:
Do you have a network firewall enabled that could be blocking the port?

ForresterLu, 2020-07-28:
> Do you have a network firewall enabled that could be blocking the port?

Hello @lemketron , I just closed the firwall and retried again, the issue remains the same. So strange!

EricBoiseLGSVL, 2020-07-28:
@ForresterLu That is odd. After changing the host name / port are you running netstat and seeing the change?  Is there any other Windows security features you can disable?  We have had another user have issues with Windows blocking connections.

ForresterLu, 2020-07-29:
> @ForresterLu That is odd. After changing the host name / port are you running netstat and seeing the change? Is there any other Windows security features you can disable? We have had another user have issues with Windows blocking connections.

@EricBoiseLGSVL Thanks for your reply, looks like I have located the problem, when the api_hostname in `config.yml` is set as `localhost`, I ran `netstat -ab` and can see the simulator was using port 8181, see below:
**config.yml**
[image]
**netstat -ab**
[image]

But when I changed `localhost` to `192.168.1.55` in `config.yml` and ran `netstat -ab`, I the simulator was using two unfamiliar ports which I did not set.
**config.yml**
[image]
***netstat -ab*
[image]

It confuses me a lot, any idea about the issue?

EricBoiseLGSVL, 2020-07-29:
@mark-gerow-lge Do know why this is happening?

martins-mozeiko, 2020-08-01:
These two are completely unrelated ports to api_hostname setting in config. These are connections (outgoing), not listening ports (incoming). Look for ports that are in `LISTENING` state and ignore everything that is in `ESTABLISHED` state.

Easiest way to guarantee connection is to put asterix in config file: `api_hostname: "*"` This will allow accept connections on any network interface - doesn't matter localhost or external.

Make check your Windows Firewall settings - that it does not block simulator.exe executable for listening on 192.168.1.0/24 interface.

closed 2020-12-12 by lemketron