lgsvl/simulator #870: Bulid LGSVL: how to include Mono support for both Windows when installing Unity?

ForresterLu, 2020-07-21:
When I build LGSVL with Unity, I saw in the bulid document [here](https://www.lgsvlsimulator.com/docs/build-instructions/), one step is `**IMPORTANT** include Mono support for both Windows and Linux when installing Unity`. But I cannot find where to select **Mono** in Unity Hub on Windows when insstall **Unity 2019.3.3f1**, the screenshot is shown here:
[image]

Can anyone know how to include Mono support for both Windows when installing Unity?

lemketron, 2020-07-21:
[image]
It's actually called "Linux Build Support (Mono)".

ForresterLu, 2020-07-21:
OK, Thank you!

closed 2020-07-21 by ForresterLu