lgsvl/simulator #871: Can I controll the environment by LGSVL APIs when the vehicle in LGSVL is controlled by Apoll?

ForresterLu, 2020-07-22:
Hello everyone, I sucessfully run the LGSVL with Apollo, then I need to control the ECO vehicle in LGSVL by APIs. 
Here[manually runtime control in LGSVL](https://www.youtube.com/watch?v=I6VE3dem5q4), in the vedio you can see I **manually** control the weather like time of day, rain, wetness and so on when **the vehicle is controlled by Apollo**. What I want to do is just like the vedio demonstrates, but the key different is I need to **do the environemnt control by APIs** shown [here](https://www.lgsvlsimulator.com/docs/python-api/)

Can anyone have an idea about how to do that?
Actually this question I havr asked before in #867 , at that time, there wre no vedio the illustrate my ideas.

ntutangyun, 2020-07-22:
The [docs](https://www.lgsvlsimulator.com/docs/python-api/) is quite clear about what to do step by step.

ForresterLu, 2020-07-22:
@ntutangyun OK, thanks for your advice.

EricBoiseLGSVL, 2020-07-22:
@ForresterLu Yes, we have API control for environment controls.  Please see PythonAPI/quickstart/18-weather-effects.py and 19-time-of-day.py to see how to do this.  Hope this helps.

closed 2020-07-27 by ForresterLu