lgsvl/simulator #872: How to control ego vehicle over ROS without AD stack

amirtakouh, 2020-07-22:
Hi,

Is it possible to control an ego vehicle without using AD stacks? Is there a topic and control command message type that could be used for this?
For example, does LGSVL subscribe to any topic which receives lgsvl_msgs/msg/VehicleControlData.msg?

Best,

dr563105, 2020-07-22:
Yes it is possible to use just ros. You can read this tutorial - https://www.lgsvlsimulator.com/docs/lane-following/. 
However, this documentation is outdated. For the current simulator version, you would need to use a sensor plugin - https://github.com/lgsvl/LaneFollowingSensor and use unity to build the project again. 

Then you can use the lanefollowing repo to run the simulation, control the vehicle.

amirtakouh, 2020-07-24:
Thank you for your response!
To be more specific, is there a direct control topic and ROS (not ROS2) msg to control the vehicle in the simulator?
Also if there is a topic and ROS msg published by the simulator reporting the vehicle status? such as speed, steering angle, gear and so on?

dr563105, 2020-07-24:
You can use I think Canbusdata topic. It has all the info you need. However you need to modify the simulator code to use ROS 1, as they wrote it for ROS2(or you don’t need to). Try it out. If you’re familiar with Unity, you can make your own sensor plugin with your requirements. LanefollowingSensor and ComfortSensor present in LGSVL github repo can give you a start. Especially LanefollowingSensor plugin has a topic for control vehicle and publishes that info. You can modify it to fit your needs.

amirtakouh, 2020-07-28:
Thank you for the tips! I will try them out.
I tried controlling the vehicle over /vehicle_cmd by publishing autoware_msgs/VehicleCmd.msg and I realized that the only part of the msg that has any effect is the twist_cmd field. Now the twist_cmd takes care of acceleration and steering however it does not cover other functionalities, such as changing the gear.
Looking at the code at https://github.com/lgsvl/simulator/blob/eda3f5941c9ab375fd3832eb351e5ae0b0bf9457/Assets/Scripts/Bridge/Ros/RosConversions.cs#L537
It seems like there is support for some other fields of the VehicleCmd msg, however when running it, only the twist_cmd field is taken into account.
Can you please shed some light on this?

dr563105, 2020-07-28:
Yes it is true that only twist_cmd has effect. I think this was raised as an issue some time ago. 
As I see in this issue -- https://github.com/lgsvl/simulator/issues/721, they are working on it. They can give you a precise answer.