lgsvl/simulator #873: Increasing the NPC traffic infront/adjacent of the ego vehicle

dr563105, 2020-07-22:
Hey team!

I would like to increase the NPC traffic around the ego vehicle. I read in this issue -- https://github.com/lgsvl/simulator/issues/843#issuecomment-654955329, that it is done intentionally. 

I'm not using PythonAPI. So I would like to edit the code directly in Unity and build the simulator. 

My requirements -
1. Increase the number of NPCs around the EGO and constantly present. I use SF map. 
2. Increase the threshold of cars waiting at the signal. 

I use 2020.03 version as the 2020.05 exposure level is too low and dark for EGO to detect/follow a lane. 

It would be great if you can please list the steps I need to do, to accomplish the task. Apart from building the simulator, adding files in Unity, I have little experience in Unity. 

Thank you.

EricBoiseLGSVL, 2020-07-22:
@dr563105 Let me see if I can help with this.
There are a couple of places we need to edit in code and scene objects
- MapOrigin.  This has options for maps to increase NPCs.  We max this value because of fps issues.  You can increase this in code if you want.  I suggest the other changes first before changing the max
[image]

- Next is NCPManager.cs.  Here you can remove the check for Spawnable (This checks to see if it is an edge lane of the map.  You can remove this so it uses any lane within the spawn area set in MapOrigin that follows the ego)
- Finally you can remove the isVisible check.  This checks if any sensors "see" the npc when it tries to spawn and aborts if it will be visible on spawn.
[image]

dr563105, 2020-07-23:
Thanks Eric. I first tried with changes in NPCManager.cs, though it had increase in NPC, there were areas where there were none. Only on changing lanes, turning at an intersection, I could make the spawn happen. Sometimes that was also not happening. 

Is it possible to make more NPCs do the changing lane frequently? 

Then I increased the NPC max count in SF map origin. It originally had `20`. I raised it to `30` which is the max limit.  Tested it again. I couldn't see much difference in traffic. In certain areas in the map, certainly get more traffic than others. 
Then I changed the NPC max count in [here](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Map/MapOrigin.cs#L56). I couldn't see an improvement. 

That count is it map related or the change I did is correct? 

I'm testing for object detection and without a lot of NPCs, the NN can't learn from the images. These changes give me an improvement from the default case.

EricBoiseLGSVL, 2020-07-23:
No problem.  Lane change is in the NPC controller code and I don't recommend changing this.  You could try to get a random index that is lane.worlpoints.length - 1 in NPCManager.cs.  This would open more places to spawn on the lanes but may have odd NPC behavior.  Then, you could reduce the size of the spawn area in MapOrigin. This would try to spawn closer to the ego.  Lastly, the traffic light timings can be adjusted with the API or in the code here.
[image]

dr563105, 2020-07-23:
Thanks Eric. 

I changed the signal labels in that file. I think I'm getting good waiting time. 

Regarding changing the value, I think you meant this variable `lane.mapWorldPositions.Count` present in this line 
`var lane = mapManager.GetLane(RandomGenerator.Next(mapManager.trafficLanes.Count));` in `NPCManager.cs`. 
Which gets the value from `mapManager`. So I'm kind of lost where to initialise the value. Can you please help? 
Despite that I can say I'm getting a good NPC traffic behaviour.

EricBoiseLGSVL, 2020-07-23:
@dr563105 Close but no, the line I am referencing is this.
[image]
This could be converted to a random index instead of 0 but you need to exclude the last index or npcs will have issues with traffic signals and stop signs.  You may even want to exclude last 2 indexes if there are > 3 waypoints.  This will let multiple npcs spawn on one lane, giving you better density. Leave the var lane as is, this makes sure you are not selecting an intersection lane that will really mess up npc behavior.

dr563105, 2020-07-24:
Thanks again Eric. I will close this issue for now. When I have issues again, I will reopen it.

dr563105, 2020-08-13:
I have a new requirement - is it possible to slow the NPCs way down, easily with a few changes to the code? 

I looked into the code and did some changes to NPCController, NPCLaneController and others. I made some crude hacks to see it was really possible. It was. The vehicles slowed down enough to form an envoy around ego. However, I was hit with multiple null exceptions.

EricBoiseLGSVL, 2020-08-14:
@dr563105 I think the only way would be to adjust the speed limit on the mapLane annotation.  Sorry, we have not made this for random traffic, only API

dr563105, 2020-08-15:
I thought so. Thanks. I finally got the hang of driving alongside NPC matching its pace. Some glorious crashes, some mad NPC driving and lots of fun.

closed 2020-08-15 by dr563105