lgsvl/simulator #874: Invalid : Unknown Sensor type VelodyLidar in Source Code

arslansarp, 2020-07-23:
Hi, I am trying to use LGSVL from source file. 

I tried to add VLC32 Lidar but It didn't work. 

https://www.lgsvlsimulator.com/docs/lidar-plugin/#velodyne-lidar-sensor-plugins . I took the exact sample json here.

[image]
[image]

However, If I download the build from github, it works with the same json.

dr563105, 2020-07-23:
Yes this behaviour is because there is no compiled velodyne sensor inside asset bundles folder of your custom build. In the build from github, you can inside asset bundles, there will be a velodyne sensor zip file. Copy that and put it into your custom build's asset bundle. Then try. It will work.

arslansarp, 2020-07-24:
Thanks it worked. But is there any possibility of using it in Unity Editor instead of build version ?

closed 2020-07-24 by arslansarp

rongguodong, 2020-07-24:
To use it in Editor, you can create a folder named "AssetBundles" at the same level of "Assets", and a sub-folder named "Sensors" inside it. Put the asset bundle "sensor_VelodyneLidarSensor" in it. (Basically having the same folder structure as for the built binary)