lgsvl/simulator #875: Can I get the reference points on the map?

yumianhuli2, 2020-07-24:
Can I get the reference points on the map directly or indirectly？And How can I DO this？

ntutangyun, 2020-07-24:
what do you mean by reference points?

EricBoiseLGSVL, 2020-07-24:
Yes, what do you mean "reference points"?  HDMap? What do you plan on doing with these points?

yumianhuli2, 2020-07-25:
> what do you mean by reference points?
The scene looks like this：
https://www.bilibili.com/video/BV1ob411g7k8?from=search&seid=12094902156495632642
There are two inputs:

One: Perceived lane lines

Two: The waypoint on the lane line：like the purple dots shown below
[image]

The purple dot on the driveway, could the LGSVL simulator give me that?
Or can I get it from the simulator in some other indirect way?

Thank you so much！

EricBoiseLGSVL, 2020-07-25:
@yumianhuli2 we are working on a lane line detection sensor and hope to release it soon.  In the meantime you can write logic that parses the closest MapLanes in simulator and create geometry or some type of UI elements in world space.

yumianhuli2, 2020-07-25:
> @yumianhuli2 we are working on a lane line detection sensor and hope to release it soon. In the meantime you can write logic that parses the closest MapLanes in simulator and create geometry or some type of UI elements in world space.

May I understand that this sensor can output the purple points in the figure above and convert them to the algorithm module in the local coordinate system? 
Thank you！

yumianhuli2, 2020-07-25:
> > @yumianhuli2 we are working on a lane line detection sensor and hope to release it soon. In the meantime you can write logic that parses the closest MapLanes in simulator and create geometry or some type of UI elements in world space.
> 
> May I understand that this sensor can output the purple points in the figure above and convert them to the algorithm module in the local coordinate system?And this module is based on visual or map?
> Thank you！

EricBoiseLGSVL, 2020-07-28:
@yumianhuli2 Yes, something very similar to this but it won't release for a bit.  You can create a custom sensor in the meantime and pull map data from MapManager.cs.  Yes, you can get local and world coordinates from mapLanes and mapLines.  It is based on both, they should be the same thing.

closed 2020-07-28 by yumianhuli2