lgsvl/simulator #876: How To Run a Scenario or Test Case with Apollo

ForresterLu, 2020-07-24:
Hello everyone, I have deployed apollo-5.0 and LGSVL on my Ubuntu 18.04, then I try to use PythonAPI to do some experiment, when I want to run a scenario or test case with apollo following the instruction [here](https://www.lgsvlsimulator.com/docs/api-how-to-run-scenario/), some error occurred and the screenshot is below:

[image]

It seems the question is the connection, but I don't konw how to solve it, anyone know how to solve it?

EricBoiseLGSVL, 2020-07-24:
@rongguodong Do you know what could be the issue?  Port? Dreamview not initialized right?

lemketron, 2020-07-24:
It looks like Dreamview is working but you could load localhost:8888 in your browser to confirm. The error message is actually from the cyber bridge, not from Dreamview, and indicates a bad request sent to the cyber bridge. This can happen if the Simulator connects to the bridge thinking it is a ROS bridge instead of a Cyber bridge.

What have you selected (and entered) in the Simulations, Maps & Vehicles tab for the vehicle and bridge info?
Is Apollo running on the same host as the Simulator?

rongguodong, 2020-07-25:
I will run this script locally to see if I can reproduce the issue.

ForresterLu, 2020-07-25:
> It looks like Dreamview is working but you could load localhost:8888 in your browser to confirm.
> 
> The error message is actually from the cyber_bridge, not from Dreamview, and indicates a bad request sent to the cyber_bridge. This can happen if the Simulator thinks it is connecting to a ROS bridge instead of a Cyber bridge.
> 
> What have you selected (and entered) in the Simulations, Maps & Vehicles tab for the vehicle and bridge info?
> Is Apollo running on the same host as the Simulator?

Hey @lemketron , my dreamviewer is working and you can see below.
[image]

The simulator setting is:
[image]
[image]
[image]

rongguodong, 2020-07-25:
It seems you are using simulator with Python API, right? Then you need select "API Only" in the 3rd image.

ForresterLu, 2020-07-25:
> It seems you are using simulator with Python API, right? Then you need select "API Only" in the 3rd image.

Hey @rongguodong , I selected "API Only" model and modified the connection port from `9090` to `8181` for API connection in `VF_S_25_Slow.py`:
`sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)`
And the script ran without error! Thanks for your advice!

lemketron, 2020-07-25:
Right, sorry for the confusion with all these ports! 

As you discovered, port 9090 is where the simulator sends sensor data to the Cyber Bridge in Apollo, but the python script needs to talk to the simulator on port 8181 (or whatever port is specified in the config.yml file). 

So it makes sense if the python script was sending API commands to port 9090 then the Cyber bridge would not understand them.

closed 2020-07-25 by lemketron