lgsvl/simulator #877: Get  points  from OpenDriveMapImporter class

yumianhuli2, 2020-07-25:
Can I get the (left/right)points (x,y) on the lane from the OpenDriveMapImporter class?Thanks!

luqiang21, 2020-07-25:
@yumianhuli2 What do you mean to get points of lane from that class? Can you clarify?

yumianhuli2, 2020-07-26:
> @yumianhuli2 What do you mean to get points of lane from that class? Can you clarify?
points on a lane line

luqiang21, 2020-07-26:
@yumianhuli2 Once you import OpenDrive map successfully, you can get lane line points through the imported map object.

yumianhuli2, 2020-07-27:
> @yumianhuli2 Once you import OpenDrive map successfully, you can get lane line points through the imported map object. For any lane, left and right boundary lines and the actual lane are created, you can find all of them under the map object.

I can get referenceLinePoints from ImportRoads() function from OpenDriveMapImporter,And then I want to get the width based on these points, what could I do？Thanks!

luqiang21, 2020-07-27:
@yumianhuli2 I see what you are asking. [here](https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Editor/OpenDriveMapImporter.cs#L871) is the line where we are computing width: `widthValue`.

yumianhuli2, 2020-07-28:
> @yumianhuli2 I see what you are asking. [here](https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Editor/OpenDriveMapImporter.cs#L871) is the line where we are computing width: `widthValue`.

Hello！Can I think of this' widthValue 'as adding or subtracting from a referencepoint to a point on the left and right boundary?

luqiang21, 2020-07-28:
@yumianhuli2 Yes, the width value is used to compute boundary line points' positions as what you show in the figure. But the normal direction for each point might be inaccurate because we are downsampling points when importing the reference line.  More detailed definitions you can refer OpenDrive format specification [link](http://opendrive.org/docs/OpenDRIVEFormatSpecRev1.4H.pdf).

yumianhuli2, 2020-07-28:
> @yumianhuli2 Yes, the width value is used to compute boundary line points' positions as what you show in the figure. But the normal direction for each point might be inaccurate because we are [...]

谢谢，陆强兄！我现在的需求主要是构造输入给算法的参考线上点的结构体，有（Point3D xyz 地图坐标系下的路径点横纵竖坐标，PointLLH 地球坐标系下路径点的经纬度和高度，EulerAngles 地图坐标系下路径点的俯仰翻滚偏航角，路径点在对应车道上的累计位置s以及路径点的曲率）。这些能从目前模拟器的资源中去直接或者间接的获取吗？

Thank you, Brother Lu Qiang! I now demand is mainly structural input to the algorithm structure of online point of reference, there are (Point3D xyz map coordinates the path to the point of transverse and longitudinal vertical coordinates, PointLLH path point coordinates of longitude and latitude and altitude, earth EulerAngles map coordinates path point pitch roll yaw Angle, path points in the corresponding lane s cumulative position and the curvature of the path points). Can these be obtained directly or indirectly from the resources of the current simulator?

luqiang21, 2020-07-28:
@yumianhuli2 不客气，我是这个“路“哈。目前模拟器的车道线和边界线都只存了离散后的点的三维坐标，也就是你的Point3D xyz，你需要PointLLH的话，得用MapOrigin间接转换。路径点的EulerAngles，累计位置s以及曲率并没有。s的话OpenDRIVE本来就有， 你可以修改一下importer保留下来， 曲率应该是需要计算一下。EulerAngles我不太确定OpenDRIVE原地图中有没有，可能也需要你自己计算了。

@yumianhuli2  You're welcome. Currently, lanes/lines only keep discretized 3D coordinates for each point. That is the same as Point3D xyz in your requirement. For PointLLH, you need to obtain yourself using MapOrigin indirectly. We don't have EulerAngles, s and curvature for each path point. For s，you can modify current importer to keep it while importing from OpenDRIVE. Curvature you can obtain based on original reference line equations in OpenDRIVE. I am not sure about EulerAngles, you may need to compute yourself.

closed 2020-07-28 by yumianhuli2

ravisoni1992, 2021-05-14:
> > @yumianhuli2 What do you mean to get points of lane from that class? Can you clarify?
> > points on a lane line（lane left points and lane right points）

Hi, have you got to know how to get the X,y points for lane?
can you please help me?