lgsvl/simulator #878: About lanelet2 map

yumianhuli2, 2020-07-25:
Hello! Is OSM a Lanelet2 map? When I import the map. osm generated from the official osM website and enter the simulator, the Console  prompted "Successfully imported Lanelet2 HD Map!" But can not see any result:)???

luqiang21, 2020-07-25:
@yumianhuli2 Lanelet2 is a format based on OSM. It has added more tags. If no error, check `View All` in `Map Annotation` window. Defaultly, map annotation is not shown.

yumianhuli2, 2020-07-26:
> @yumianhuli2 Lanelet2 is a format based on OSM. It has added more tags. If no error, check `View All` in `Map Annotation` window. Defaultly, map annotation is not shown.

HI！Importing Opendrive and using View All is correct, OSM is wrong

luqiang21, 2020-07-26:
@yumianhuli2 That's expected. We only support importing lanelet2 not OSM.

closed 2020-07-27 by yumianhuli2