lgsvl/simulator #879: Executable File (simulator) Cannot Be Executed On Ubuntu

VentiOvaltine, 2020-07-26:
The only output in the shell is:
```
Set current directory to /mnt/c/Users/chong/Dev/lgsvlsimulator-linux64-2020.05
Found path: /mnt/c/Users/chong/Dev/lgsvlsimulator-linux64-2020.05/simulator
```
If double click `simulator`, a window pop up and suddenly closed. I don't have a Nvidia GPU on my machine, is that cause this issue?
And how can I open the simulator? Thans!

dr563105, 2020-07-27:
Do you any other GPU in your machine? Also it will help you can attach the simulator player log. You can find the location of the log in the faq of the lgsvl website.

EricBoiseLGSVL, 2020-07-28:
@VentiOvaltine Yes, we will need more information to debug this issue.  It seems that it is a config or hardware issue for a sudden crash on start.

ChihweiLHBird, 2020-07-29:
> Do you any other GPU in your machine? Also it will help if you can attach the simulator player log. You can find the location of the log in the faq of the lgsvl website.

I was working with @VentiOvaltine , and we were in a virtual machine where AMD GPU driver was not installed. The AMD GPU is RX 5700.

EricBoiseLGSVL, 2020-07-30:
Yes, you will need a GPU driver installed.  We have a few other issues from other users using VM's.

ChihweiLHBird, 2020-07-30:
> Yes, you will need a GPU driver installed. We have a few other issues from other users using VM's.

Thank you, so the recommendation would be using Ubuntu on a real machine, right?

EricBoiseLGSVL, 2020-07-30:
Yes, it is much easier

bobbyhaliwela, 2020-08-06:
I am having the same issue here, difference is i'm not using a virtual machine. 

System: Ubuntu 18.04.4 LTS
Memory: 15,5 GiB
Processor: Intel® Core™ i7-4790 CPU @ 3.60GHz × 8
Graphics: GeForce GTX 1070 Ti/PCIe/SSE2

Followed each steps on [https://www.lgsvlsimulator.com/docs/getting-started/#getting-started](url). Double clicking the executable or execute `./simulator` from terminal does not work. A window pop up for a second then suddenly closed. And `libvulkan1` is already installed.

`nvidia-smi` output:

```
> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 450.57       Driver Version: 450.57       CUDA Version: 11.0     |
> |-------------------------------+----------------------+----------------------+
> | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
> |                               |                      |               MIG M. |
> |===============================+======================+======================|
> |   0  GeForce GTX 107...  Off  | 00000000:01:00.0  On |                  N/A |
> |  0%   53C    P8     9W / 180W |    441MiB /  8118MiB |      0%      Default |
> |                               |                      |                  N/A |
> +-------------------------------+----------------------+----------------------+
>                                                                                
> +-----------------------------------------------------------------------------+
> | Processes:                                                                  |
> |  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
> |        ID   ID                                                   Usage      |
> |=============================================================================|
> |    0   N/A  N/A      1543      G   /usr/lib/xorg/Xorg                 26MiB |
> |    0   N/A  N/A      1605      G   /usr/bin/gnome-shell               55MiB |
> |    0   N/A  N/A      1803      G   /usr/lib/xorg/Xorg                200MiB |
> |    0   N/A  N/A      1935      G   /usr/bin/gnome-shell              137MiB |
> |    0   N/A  N/A      2604      G   /usr/lib/firefox/firefox            5MiB |
> |    0   N/A  N/A      3377      G   /usr/lib/firefox/firefox            5MiB |
> |    0   N/A  N/A     12882      G   gnome-control-center                5MiB |
> +-----------------------------------------------------------------------------+
```

lemketron, 2020-08-06:
> Followed each steps on [https://www.lgsvlsimulator.com/docs/getting-started/#getting-started](url). Double clicking the executable or execute `./simulator` from terminal does not work. A window pop up for a second then suddenly closed. And `libvulkan1` is already installed.

@bobbyhaliwela Could you take a look at (or post here) the Player.log file from ~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log as there may be some clues in that log file.

bobbyhaliwela, 2020-08-06:
> @bobbyhaliwela Could you take a look at (or post here) the Player.log file from `~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log` as there may be some clues in that log file.

No such file or directory. Did i miss something?

bobbyhaliwela, 2020-08-06:
> @bobbyhaliwela Could you take a look at (or post here) the Player.log file from `~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log` as there may be some clues in that log file.

My mistake before. Here's the output:

```
> Mono path[0] = '/home/bob/lgsvlsimulator-linux64-2020.05/simulator_Data/Managed'
> Mono config path = '/home/bob/lgsvlsimulator-linux64-2020.05/simulator_Data/MonoBleedingEdge/etc'
> Preloaded 'liblaszip.so'
> Preloaded 'libpng.so'
> Preloaded 'libsqlite3.so'
> Preloaded 'libturbojpeg.so'
> Display 0 'DELL S2415H 24"': 1920x1080 (primary device).
> Display 1 'DELL S2415H 24"': 1920x1080 (secondary device).
> Loading player data from /home/bob/lgsvlsimulator-linux64-2020.05/simulator_Data/data.unity3d
> Desktop is 1920 x 1080 @ 60 Hz
> [Vulkan init] extensions: count=18
> [Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
> [Vulkan init] extensions: name=VK_KHR_display, enabled=1
> [Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
> [Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
> [Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
> [Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
> [Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
> INTEL-MESA: warning: Haswell Vulkan support is incomplete
> Vulkan detection: 2
> Initialize engine version: 2019.3.3f1 (7ceaae5f7503)
> [Subsystems] Discovering subsystems at path /home/bob/lgsvlsimulator-linux64-2020.05/simulator_Data/UnitySubsystems
> GfxDevice: creating device client; threaded=1
> [Vulkan init] extensions: count=18
> [Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
> [Vulkan init] extensions: name=VK_KHR_display, enabled=1
> [Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_surface, enabled=1
> [Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
> [Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
> [Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
> [Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
> [Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
> [Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
> [Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
> [Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
> INTEL-MESA: warning: Haswell Vulkan support is incomplete
> [Vulkan init] extensions: count=65
> [Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
> [Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
> [Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
> [Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
> [Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
> [Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
> [Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
> [Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[... 85 lines cut ...]
> found manifest at Controllables/controllable_TrafficCone/manifest 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Fallback handler could not load library /home/bob/lgsvlsimulator-linux64-2020.05/simulator_Data/Mono/data-0x2df00c0.so
> WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
> WARNING: Shader Did you use #pragma only_renderers and omit this platform?
> found manifest at Sensors/sensor_CheckControlSensor/manifest 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Fallback handler could not load library /home/bob/lgsvlsimulator-linux64-2020.05/simulator_Data/Mono/data-0x2d7ea40.so
> found manifest at Sensors/sensor_VelodyneLidarSensor/manifest 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> Fallback handler could not load library /home/bob/lgsvlsimulator-linux64-2020.05/simulator_Data/Mono/data-0x2d857e0.so
> loaded 3 NPCs behaviours and 6 NPC models in 00:00:00.1389268 
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)
> 
> X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
>   Major opcode of failed request:  149 ()
>   Minor opcode of failed request:  4
>   Resource id in failed request:  0x260000f
>   Serial number of failed request:  271
>   Current serial number in output stream:  277
> 
```

dr563105, 2020-08-06:
You can see in the player.log that your graphic card is not enabled. The simulator can only see Intel graphics `INTEL-MESA: warning: Haswell Vulkan support is incomplete`. 

Using at this part we get the GPU name. Something like this. 
```
Vulkan API version 1.1.0 (1.1.119 supported by driver)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce RTX 2080 Ti] id=[1e07]
Vulkan device type 2
Vulkan driverversion=[0x6e190000] uint=[1847132160]
Vulkan PSO: cache data not found [path=/tmp/LG Silicon Valley Lab/LGSVL Simulator/vulkan_pso_cache.bin]
```
But you got this 
```
Vulkan API version 1.1.0 (1.2.131 supported by driver)
> Vulkan vendor=[Intel] id=[8086]
> Vulkan renderer=[Intel(R) HD Graphics 4600 (HSW GT2)] id=[412]
> Vulkan device type 1
> Vulkan driverversion=[0x05000008] uint=[83886088]
> Vulkan PSO: cache data not found [path=/tmp/LG Silicon Valley Lab/LGSVL Simulator/vulkan_pso_cache.bin]
```

One more thing you can try is having `nvidia-smi` open while you open the simulator. Check if the memory spikes or even recognised.

EricBoiseLGSVL, 2020-08-06:
@bobbyhaliwela Yes, @dr563105 is correct.  You must use a vulkan supported GPU, Intel graphics will not work.  Be sure to search Issues for help on making sure vulkan is installed correctly.

lemketron, 2020-08-07:
From https://github.com/lgsvl/simulator/issues/287#issuecomment-613210044:

> In particular, if you see "mesa" or "Intel" mentioned in Vulkan vendor or renderer lines rather than NVIDIA, then you might have "mesa-vulkan-drivers" installed. If so, then you can remove these with `sudo apt remove mesa-vulkan-drivers`.
> ... Type `vulkaninfo` to verify that libvulkan is working, and then see if `vkcube` displays a window with a spinning 3D cube.

Then try `nvidia-smi` again to confirm the nvidia drivers are still installed and recognizing your Nvidia 1070 GPU, and finally try the simulator again.

bobbyhaliwela, 2020-08-07:
> From [#287 (comment)](https://github.com/lgsvl/simulator/issues/287#issuecomment-613210044):
> 
> > In particular, if you see "mesa" or "Intel" mentioned in Vulkan vendor or renderer lines rather than NVIDIA, then you might have "mesa-vulkan-drivers" installed. If so, then you can remove these with `sudo apt remove mesa-vulkan-drivers`.
> > ... Type `vulkaninfo` to verify that libvulkan is working, and then see if `vkcube` displays a window with a spinning 3D cube.
> 
> Then try `nvidia-smi` again to confirm the nvidia drivers are still installed and recognizing your Nvidia 1070 GPU, and finally try the simulator again.

This solved the issue. Thanks!

closed 2020-08-07 by EricBoiseLGSVL