lgsvl/simulator #880: Turn off default setting

yumianhuli2, 2020-07-27:
Entering the scene will generate the following elements by default

[image]

How can I turn this setting off?Thank you！

EricBoiseLGSVL, 2020-07-28:
@yumianhuli2 This is functionality that is needed for the simulation to run, this cannot be turned off.  What are you trying to do?

yumianhuli2, 2020-07-28:
> @yumianhuli2 This is functionality that is needed for the simulation to run, this cannot be turned off. What are you trying to do?

Hello,I want to use the resources of the simulator to create a scene that meets my personal needs.So please tell me how to turn off these defaults, thank you！

EricBoiseLGSVL, 2020-07-28:
@yumianhuli2 Again, these are required to run the simulator.  HDRP requires PostProcessing, Lights.  Managers are required for the functionality.  You can create your own scene but there are required objects that need to be added or created at runtime.  Are you just asking how to make your own environment?  If so, please see [this](https://youtu.be/S0w2ahhEPbE).

yumianhuli2, 2020-07-29:
> @yumianhuli2 Again, these are required to run the simulator. HDRP requires PostProcessing, Lights. Managers are required for the functionality. You can create your own scene but there are required [...]

Hi！In short, I just want an empty scenario that contains the necessary elements of the HDRP.
Thanks for reply!

closed 2020-08-28 by EricBoiseLGSVL