lgsvl/simulator #881: Unable to receive Segmentation camera images in a subscriber node

dr563105, 2020-07-27:
Hey everyone,

I'm trying to capture segmentation camera images and I'm failing at it in the subscriber node. I use ros2 and python. 
In my subscriber node, I subscribe to the segmentation camera topic `'/simulator/segmentation_camera'` with message type as `CompressedImage`. Then in the callback, I just save them using cv2 library. 

Just like [radar and ground truth 3d detections](https://github.com/lgsvl/simulator/issues/854#issuecomment-663908101), I can't receive them. I can see what the sensors see in the visualiser window. 

I'm using colour and depth camera; they work. 

What am I doing wrong and can you please help? 

Thank you.

EricBoiseLGSVL, 2020-07-28:
@rongguodong Can you look at this issue when you can please? Thanks

rongguodong, 2020-07-28:
@dr563105 Can you share your Python script to show what you are doing? I tested color camera, depth camera and segmentation camera on ROS2, and did not see any difference among them.
If you are fine with the former two but only get the error with segmentation camera, what error do you get? Please share more details on how to reproduce it.

dr563105, 2020-07-28:
Sure. 

I use the lanefollowing docker image and its `docker-compose up collect` to run the subscriber node. 
However, you can try it outside of the docker. The docker image uses python3.6.9, ROS2 dashing

The problem is I'm not getting segmented camera images at the subscriber. I can store colour camera and depth camera images. 
Also I'm not getting any error. Neither in terminal nor in the Simulator. 

Additionally, I tested if I indeed receive the topic by going into the docker and executing `ros2 topic list` and `ros2 topic echo /simulator/sensor/segmentation_camera`, no output. But,  I get outputs for colour and depth images. 

I don't know why. 

Attaching the script(remove txt extension and add .py)
[collect.txt](https://github.com/lgsvl/simulator/files/4990297/collect.txt)
 and sensor parameter json file. 
[sensor_parameters_json.txt](https://github.com/lgsvl/simulator/files/4990288/sensor_parameters_json.txt)

P.S - If you face `null reference exception` error for radar sensor, please edit/add the line as suggested [here](https://github.com/lgsvl/simulator/issues/854#issuecomment-662657702) and [here](https://github.com/lgsvl/simulator/issues/854#issuecomment-663908101) or simply remove radar from sensor parameters.

Thanks.

rongguodong, 2020-07-28:
I just tested "rost2 topic echo", and I can see both color camera and segmentation camera. No difference.

One possible thing: are you using "ROS2 Web Bridge" (https://www.lgsvlsimulator.com/docs/autoware-auto-instructions/#install-ros2-web-bridge)? It is known to have performance issue and cannot handle too many sensors (particularly if you have multiple cameras and lidars). So you may want to try only using segmentation camera and remove others sensor, to see if you are able to see something using "ros2 topic echo" for segmentation camera.

dr563105, 2020-07-28:
I already tried just segmentation camera. Not luck. Callback isn't even invoked. Tried only with print statement in callback. It doesn't get executed meaning the subscribed topic is not receiving data.

I use the data collect script in [lanefollowing](https://github.com/lgsvl/lanefollowing). I used [docker-compose](https://github.com/lgsvl/lanefollowing/blob/master/docker-compose.yml#L16) file invoke [collect script](https://github.com/lgsvl/lanefollowing/blob/master/scripts/collect.sh) where [`rosbridge`](https://github.com/lgsvl/lanefollowing/blob/master/scripts/ros2_bridge.sh) is called. 

I haven't yet tried ROS2 Web bridge method. Although I must admit, the `rosbridge` process occupies 90% CPU in one or two cores.  So far I have the collect script multiple times for data collection of images, it has worked fine. I now add segmentation camera topic to it, I don't receive any. 

Tested again with built 2020.05 binary from release page. No luck.

daviduhm, 2020-07-28:
@dr563105 I've tried to reproduce it with 2020.05 and the ros2 bridge in lane following docker image, and it works as below:

[image]
[image]

Can you try with `ros2 topic echo` and see if image messages are coming from the simulator side? Without knowing how exactly you've added segmentation camera topic to the collect script, it's hard to tell from our side if the script itself is working fine.

As you might already know, the script uses `ApproximateTimeSynchronizer` for multiple topics and it'll only invoke a callback function when all subscribers are receiving messages.

daviduhm, 2020-07-28:
@dr563105 I didn't pay attention that you've already included your script up above. I've also tested your collect script with segmentation camera and color camera and was able to get the callback function getting called with the log messages as below:

[image]

dr563105, 2020-07-28:
Thanks David. 

I don't know why the script doesn't work for me. 
This is how I get the output when I execute `ros2 topic echo /simulator/segmentation_camera` in the terminal. 

[image]

And how the terminal looks like when `docker-compose up collect` output looks like - 
[image]

The `callback` isn't getting invoked. 

All I did was replace [this](https://github.com/intel/ros2_message_filters/tree/e94eebd0174e1d63069fa8ff36c9833c5c39cb26) with the new ros2 [message_filter](https://github.com/ros2/message_filters/tree/dashing). Then I added lgsvl_msgs from the repo. I see the exact behaviour with radar messages. 

Do you know how to debug the rosbridge while the docker is running? I would like to see the logs.

dr563105, 2020-07-28:
I just tried  outside of the docker and did `DEBUG=ros2-web-bridge* node bin/rosbridge.js 2> log.txt`. I can see segmentation_camera output. 

Could it be that, since I list all the sensors and radar. There is a bottleneck? I will try tomorrow if I can do one by one in the sensor parameters in web ui. Quite weird that you can get the output. 

If this is the case, how can I sync different sensor info under one timestamp?

rongguodong, 2020-07-28:
You have 7 cameras? Are they all 1920x1080? That should be a problem for ROS2 Web Bridge.

But you said above that even if you only have one single segmentation camera, you still has errors. Can you confirm that?

daviduhm, 2020-07-28:
My test setup only includes one segmentation camera and one color camera as that's all required topics for your modified collect script. It's possible that there's a bottleneck in bridge if you're using too many sensors, especially Lidar or camera sensor, and I can see from your sensor JSON that you're using 7 camera sensors in total. I'd suggest to minimize the number of sensors in your sensor setup.

dr563105, 2020-07-29:
@rongguodong 

> You have 7 cameras? Are they all 1920x1080? That should be a problem for ROS2 Web Bridge.
> 
> But you said above that even if you only have one single segmentation camera, you still has errors. Can you confirm that?

I tried twice with just segmentation and colour. I didn't receive anything. However, now just like @daviduhm, I tried what the script had -- one colour and one segmentation. Still I get nothing in the terminal where the script is executed. 
But, I get outputs when I debug the rosbridge. 
Sometimes I get this error -
```
2020-07-29T07:13:19.268Z ros2-web-bridge:Bridge Exception caught in Bridge.executeCommand(): TypeError: Cannot read property 'get' of undefined
2020-07-29T07:13:19.268Z ros2-web-bridge:Bridge Error: TypeError: Cannot read property 'get' of undefined happened when executing command publish
2020-07-29T07:13:19.268Z ros2-web-bridge:Bridge Response: {"op":"set_level","level":"error"}
```
other times it's fine. 
Attaching the rosbridge log. Let me know please if you find something unique. 
[log.txt](https://github.com/lgsvl/simulator/files/4993260/log.txt)(size = 22.4MB)

dr563105, 2020-07-29:
Ok. It is working now. I reckon, it was because some conflicts with conda environment. I had to exit conda env, then build the ros2 packages and try the rosbridge. I was able to receive the data. 
I then proceeded to test with lanefollowing docker. It works. 

My goal was to collect as much high quality images as possible and then do post processing where necessary. I fear you're right @rongguodong. I have since reduced the dimensions to 640x480. Let me see if this works. 

Thanks for your help. @rongguodong @daviduhm

closed 2020-07-29 by dr563105