lgsvl/simulator #882: View Rostopics list of lgsvl

ashwinsarvesh, 2020-07-27:
Hello,

I would like to run the command "rostopic list" and "rostopic echo /topicname" and probably other commands on the terminal to view the ROS topics (eg CAN bus data) but the commands are not found. I know that something has to be installed, but not sure which one
Can you help?

lemketron, 2020-07-27:
Are you using ROS (and possibly Autoware)? You might find the Autoware.Auto Lecture Series helpful:
https://www.apex.ai/autoware-course

My video for Lecture 11 was posted to YouTube last week and has a hands-on walkthrough of using the LGSVL Simulator with ROS2 and Autoware.Auto: https://youtu.be/OcB6FUbjZXo

If you are using Cyber RT and Apollo then the commands are different. We will post a similar video soon for Apollo but in the mean time you could checkout cyber_monitor and cyber_visualizer as documented here: https://github.com/ApolloAuto/apollo/blob/master/docs/cyber/CyberRT_Developer_Tools.md

ashwinsarvesh, 2020-07-27:
Hello,

Thank you for your reply. 
I am using Apollo 3.0. Can you help with that?

lemketron, 2020-07-27:
Apollo 3.0 is based on ROS, as you can see in the docs in the 3.0.0 branch of apollo here: https://github.com/ApolloAuto/apollo/blob/r3.0.0/docs/howto/how_to_run_offline_perception_visualizer.md

So the regular ROS commands should work...  You can also visualize the bridge connection status and see what topics the simulator is publishing, as documented here: https://www.lgsvlsimulator.com/docs/bridge-connection-ui/

ashwinsarvesh, 2020-07-28:
@lemketron Thank you for the bridge connection status link.
To run the basic ros commands (eg. rostopic list) in the terminal, should I install anything?

dr563105, 2020-07-28:
1. Yes you need to install ROS. Use ROS Melodic. It is stable and tested by many. 
2. Then make a rosbridge between your terminal and Simulator. If successful, you will see that in the "bridge UI" `connected` status.
3. Then set your ROS environment variables(without this you will get "commands not found" error). Remember this must be done every time you open a terminal. Else none of ROS commands work.
4. After, you should know the topic names you want to check. 
5. Run `rostopic list` to see all the topics available. Choose appropriate command for your needs. In future, you can use RViz to visualise the topics.
6. For ROS related questions, you can ask in ROS answers.
7. I assume you're new to ROS, so I suggest you spend sometime reading about ROS -- http://wiki.ros.org/

martins-mozeiko, 2020-08-01:
But don't forget that Apollo 3.0 has modified ROS. So while upstream Melodic version will work for some commands, like rostopic list. It won't work for subscribing and publishing messages that are Apollo specific type (based on protobuf). For that you'll need to use [Apollo ROS](https://github.com/ApolloAuto/apollo-platform). Easiest way to get it is to use Apollo 3.0 docker container to run ros commands/scripts.

ashwinsarvesh, 2020-08-03:
Hello,

Which means to view the data of the topic "/apollo/canbus/chassis" I cannot use the upstream Melodic version?

martins-mozeiko, 2020-08-03:
That is correct. All custom apollo topics uses protobuf messages that won't be available with ROS Melodic version. If you check type of topic with `rostopic type /topic/name` and it is in pb_msgs namespace - that mean it is protobuf format.