lgsvl/simulator #883: Apollo halting at intersections and providing erratic paths

UYasher, 2020-07-27:
I'm trying to get Apollo 5.5 to run with LGSVL following [these instructions](https://www.lgsvlsimulator.com/docs/apollo-master-instructions/) on Ubuntu 20.04.1 LTS.

While I am able to run the two pieces of software together, and the car will move, it stops at the first intersection of Borregas Ave and refuses to move forward even though the light is green. I can manually drive the vehicle into the intersection, and it will start driving autonomously again; however, the behavior becomes erratic, swerving between lanes after the slight bend in the lane after the intersection.

I've tried to follow the tutorial above as carefully as I can, but the steps I took do differ slightly:
1) I ran `./apollo.sh build` instead of `./apollo.sh build_opt_gpu` as the latter fails to compile.
2) The perception and traffic light modules automatically switch to off shortly after I turn them on. I am using the "Third Party Perception" module instead of the perception module as a result.

In Dreamview, the following also seem potentially noteworthy?:
1) I the following warning -- Raw IMU Message Delay. IMU message is 2118 cycle 21.1839 sec behind current time. (Similar for Raw GPS Message.)
2) GPS and Recorder both show "FATAL". (Lidar 128 and Localization show "OK".)

Docker Version: apolloauto/apollo:dev-x86_64-18.04-20200721_1840

I'd love to get LGSVL running with Apollo, to use for research purposes. Any idea why it might not be working?

rongguodong, 2020-07-27:
Apollo master has problems (see this issue for more details: https://github.com/ApolloAuto/apollo/issues/11818). You may want to try "pre6" branch for now.

UYasher, 2020-07-27:
Thanks rongguodong, that solved the perception issues which prevented it from moving past the intersection. It still swerves between lanes using pre-6 though.

rongguodong, 2020-07-28:
I heard that Apollo's CPU-only perception dose now work well. Not sure if it is related to your problem. Can you try "build_gpu" or "build_opt_gpu"?

You said "build_opt_gpu" fails. What errors did you get? What is your GPU? Do you have latest NVIDIA and NVIDIA-docker installed?

rongguodong, 2020-07-29:
I saw the same "swerving" on my colleague's machine. It is because he run both simulator and Apollo on a machine which is now powerful enough. As the result, the FPS of simulator is only around 10.

Please check FPS for your simulator. If it is too low, please try to run simulator and Apollo on different machines, or run both of them on a more powerful machine.

UYasher, 2020-07-30:
Using the pre-6 branch, I am able to use build_opt_gpu. The issue of swerving persists, unfortunately.

I don't think the issue is a hardware problem. My system specs (below) should be sufficient, I think?:
i7 7700K (8-core 4.20GHz)
GeForce GTX 1080 (8GB VRAM)
16GB Ram

Running `nvidia-smi -l 1` and the Ubuntu task monitor while swerving gives this:
[image]

Happy to check the FPS though, what is the best way to do that? And if you think the issue isn't hardware do you have any other suggestions?

rongguodong, 2020-07-30:
One simple way to test: Can you run simulator and Apollo on two different machines, and see if the situation is better?

UYasher, 2020-07-30:
Good idea, I'll try that!

Is the only needed change to update the connection string from localhost:9090 to the ip and port of the machine running Apollo, or are there other changes? (A guide to running on multiple machines might also be helpful.)

rongguodong, 2020-07-30:
On the client machine (where simulator is running), open WebUI and setup ip:port in "simulation" page. You should put the IP of the server machine (where Apollo and bridge.sh are running) there with the same port (9090).

UYasher, 2020-08-10:
I followed these instructions, but the behavior of Apollo and LGSVL deteriorated with a two-machine setup.

Although LGSVL and Apollo will connect, the car in Apollo will stay in place after I send a routing request. Then, Apollo will keep trying to accelerate the car, leading it to move forward wildly in LGSVL. The car then crashes in LGSVL, and eventually Apollo also gets that data, but with some delay.

I also get the following errors from Apollo's `./scripts/bridge.sh` after the crash --
E0810 16:36:37.556474 36914 client.cc:88] [cyber_bridge]Client read failed, disconnectingsystem:104
E0810 16:36:37.557931 36914 client.cc:97] [cyber_bridge]Client write failed, disconnectingsystem:32

closed 2020-08-24 by UYasher

hyt1204, 2025-04-21:
> 我按照这些说明进行作，但 Apollo 和 LGSVL 的行为在两台机器设置下恶化了。
> 
> 尽管 LGSVL 和 Apollo 将连接，但在我发送路线请求后，Apollo 中的汽车将保持在原位。然后，Apollo 将继续尝试加速汽车，带领它在 LGSVL 中疯狂前进。然后，汽车在 LGSVL 中坠毁，最终 Apollo 也获得了该数据，但有一些延迟。
> 
> 崩溃后，我还从 Apollo 那里收到了以下错误 -- E0810 16：36：37.556474 36914 client.cc:88] [cyber_bridge]客户端读取失败，断开连接系统：104 E0810 16：36：37.557931 36914 client.cc:97] [cyber_bridge]客户端写入失败，断开连接系统：32`./scripts/bridge.sh`