lgsvl/simulator #884: When click start,  causing the computer to shut down,

liukr001, 2020-07-29:
On Ubuntu 18.04, the 2060 Nvidia graphics card,

EricBoiseLGSVL, 2020-07-29:
@liukr001 Please post the player log and more information.

liukr001, 2020-07-30:
the player log shows that :
SocketException: Connection refused
  at System.Net.Sockets.SocketAsyncResult.CheckIfThrowDelayedException () [0x00014] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at System.Net.Sockets.Socket.EndConnect (System.IAsyncResult asyncResult) [0x0002c] in <ae22a4e8f83c41d69684ae7f557133d9>:0 
  at Simulator.Bridge.Cyber.Bridge.<Connect>b__25_0 (System.IAsyncResult ar) [0x00000] in <3e4812d674794809853e4a843dd3a04f>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Bridge.Cyber.Bridge:<Connect>b__25_0(IAsyncResult)
System.Net.Sockets.<>c:<Complete>b__27_0(Object)
System.Threading.QueueUserWorkItemCallback:System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
System.Threading.ThreadPoolWorkQueue:Dispatch()
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()
 
(Filename: <ae22a4e8f83c41d69684ae7f557133d9> Line: 0)
...

liukr001, 2020-07-30:
> @liukr001 Please post the player log and more information.

the player log shows that :
SocketException: Connection refused
at System.Net.Sockets.SocketAsyncResult.CheckIfThrowDelayedException () [0x00014] in :0
at System.Net.Sockets.Socket.EndConnect (System.IAsyncResult asyncResult) [0x0002c] in :0
at Simulator.Bridge.Cyber.Bridge.b__25_0 (System.IAsyncResult ar) [0x00000] in <3e4812d674794809853e4a843dd3a04f>:0
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Bridge.Cyber.Bridge:b__25_0(IAsyncResult)
System.Net.Sockets.<>c:b__27_0(Object)
System.Threading.QueueUserWorkItemCallback:System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
System.Threading.ThreadPoolWorkQueue:Dispatch()
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()

(Filename: Line: 0)
...

EricBoiseLGSVL, 2020-07-30:
It seems this may be an issue with your bridge setup.  Can you post what host and port you are trying to send to and the ports that are being used at the time please?

rongguodong, 2020-07-30:
The log shows errors of bridge connection. But that should not cause PC to be turned off.

Maybe something related to GPU and/or other hardware issue? Is your power supply powerful enough to drive GPU and everything?

liukr001, 2020-08-01:
> It seems this may be an issue with your bridge setup. Can you post what host and port you are trying to send to and the ports that are being used at the time please?

Yeah, I deleted the log file, tried again, and there was no record of the above problem in the log file. This could be a previous recording problem, not a shutdown problem. The port is IP: 9090

liukr001, 2020-08-01:
> The log shows errors of bridge connection. But that should not cause PC to be turned off.
> 
> Maybe something related to GPU and/or other hardware issue? Is your power supply powerful enough to drive GPU and everything?
The Apollo industrial computer is used, the graphics card is nvidia rtx 2060 super, and the maximum power of the power adapter is 280w. Is it necessary to set the parameters of the graphics card?

EricBoiseLGSVL, 2020-08-03:
@liukr001 After looking at a few websites on the recommended PSU for the 2060 RTX Super is around 500W.  280W is way too low

closed 2020-08-28 by EricBoiseLGSVL