lgsvl/simulator #885: Is RTX 2060 Enough for Working with Apollo and LGSVL?

ChihweiLHBird, 2020-07-29:
Is RTX 2060 GPU enough for working with Apollo and LGSVL?

EricBoiseLGSVL, 2020-07-30:
@ChihweiLHBird yes but you will need to use the Apollo 5.5 forked off master.  Apollo 5.0 does not support this GPU

ChihweiLHBird, 2020-07-30:
> @ChihweiLHBird yes but you will need to use the Apollo 5.5 forked off master. Apollo 5.0 does not support this GPU

Thank you!

ChihweiLHBird, 2020-07-30:
> @ChihweiLHBird yes but you will need to use the Apollo 5.5 forked off master. Apollo 5.0 does not support this GPU

By the way, what would be the recommended hardware for using LGSVL simulator with Apollo? It is often said that the simulator works better on a Windows machine, so is it better to have two computers (1 with Windows the simulator and 1 with Ubuntu and Apollo)?

lemketron, 2020-07-30:
> is it better to have two computers (1 with Windows the simulator and 1 with Ubuntu and Apollo)?

Yes, that is ideal, if possible.

ChihweiLHBird, 2020-07-31:
> > is it better to have two computers (1 with Windows the simulator and 1 with Ubuntu and Apollo)?
> 
> Yes, that is ideal, if possible.

And is there any GPU that's recommended or proffered?

closed 2020-07-31 by ChihweiLHBird