lgsvl/simulator #886: Lgsvl Apollo [cyber_bridge] fail to setsockopt IP_ADD_MEMBERSHIP, No such device

XuehuanYang, 2020-07-31:
Describe the bug
I implemented LGSVL-2020-03 simulation with Apollo-5.0 platform and launch LiDAR camera radar perception module.  However, it shows us the error message,  **[cyber_bridge] fail to setsockopt IP_ADD_MEMBERSHIP, No such device**

Desktop (please complete the following information):

OS: Ubuntu 18.04
Platform: Apollo-5.0
Simulation: LGSVL-2020-03
Browser: Google Chrome
Map: San Francisco
Vehicle: Lincoln2017MKZ (Apollo 5.0)

To Reproduce
Steps to reproduce the behavior:
$ ./lgsvl/lgsvlsimulator-linux64-2020.03/simulator
Check Maps -> Vehicles -> Cluster (127.0.0.1) -> Simulation -> Only Use API to control simulation.

$ cd apollo-5.0
$ sudo bash docker/scripts/dev_start.sh
$ sudo bash docker/scripts/dev_into.sh

root@in_dev_docker:/apollo# sudo scripts/bootstrap.sh
root@in_dev_docker:/apollo# sudo scripts/bridge.sh

root@in_dev_docker:/apollo# cyber_monitor

Go to http://localhost:8888/ -> Module Control -> Localization & Perception & Transformation
$ cyber_launch start /apollo/modules/perception/production/launch/perception_all.launch

Expected behavior
The command $ source scripts/bridge.sh should work fine, but it shows us the error.  Appreciate it your kind support

[image]

rongguodong, 2020-07-31:
https://stackoverflow.com/questions/3187919/error-no-such-device-in-call-setsockopt-when-joining-multicast-group#:~:text=What%20does%20%22IP_ADD_MEMBERSHIP%3A%20No%20such,t%20have%20multicast%20support%20enabled.

martins-mozeiko, 2020-08-01:
You are running Apollo as root user. Apollo does not like running as root user. Do not run "sudo dev_start.sh" or "sudi dev_into.sh". Always run without sudo.

devynliu, 2020-08-12:
### Q： 2台机器的通信链接：
#### 必要准备
1. Apollo5.0 在笔记本Ubuntu18.04上运行；
2. lgsvl simulation 运行在工控机 win10上；
#### 建立局域网，更改IP地址：
1. Ubuntu 以太网：127.17.0.4 子网掩码：255.255.255.0 	(自己设置)
2. win10上：127.17.0.5 子网掩码：255.255.255.0	(自己设置)
3. Ubuntu apollo docker上：127.17.0.1 子网掩码：255.255.0.0	(不用更改，不动。要改估计要在docker源码里改)
#### 可能出现“fail to setsockopt IP_MULTICAST_LOOP ...”
1. 解决：find multicast_notifier.cc 文件；文件中有个：std::string mcast_ip("239.255.0.100"); 组播地址；
2. 根据：[Lgsvl Apollo cyber_bridgefail to setsockopt IP_ADD_MEMBERSHIP, No such device #886](https://github.com/lgsvl/simulator/issues/886)
3. [Error “No such device” in call setsockopt when joining multicast group](https://stackoverflow.com/questions/3187919/error-no-such-device-in-call-setsockopt-when-joining-multicast-group#:~:text=What%20does%20%22IP_ADD_MEMBERSHIP%3A%20No%20such,t%20have%20multicast%20support%20enabled)
4. 查看本机的路由情况；
找到自己局域网的网卡名：enx000ec6672d1e：最后一列；
```bash
route -e 
route add -net 239.255.0.100 netmask 255.255.255.255 enx000ec6672d1e 
```