lgsvl/simulator #887: preferred system and GPU recommendations for connecting LGSVL and Apollo

VentiOvaltine, 2020-07-31:
Question1: If I would like to run both LGSVL and Apollo in **the same** system, what is the preferred system? Is there any GPU recommendations?

Question2: If I run them in parallel in **two different machines**, does this work better? If so, how to make a connection with no delay between them? 

I am really confused about the setup.

rongguodong, 2020-07-31:
For question 2: it would better to run them on two different machine (Windows for simulator and Ubuntu for Apollo). As long as the two machines are both in a local network and the network is Gibabit Ethernet, they will work well. This is how I do my daily test. :)

EricBoiseLGSVL, 2020-08-04:
@VentiOvaltine 1) As powerful as you can get.  The reason we recommend using Simulator on Windows is that Unity is better optimized for this OS.  Apollo is only on Linux. 2) As @rongguodong stated, yes.  Ports and address set in bridge to connect to the other machine.  Simulator runs and sends the bridge data to those connections instead of localhost.  Just plug both PC's into one GIGAbit (lower transfer rates will fail to run Apollo) router and you will have much better performance.

Thinsheep, 2021-03-25:
@EricBoiseLGSVL 
Hi,
Do I need Gigabit Ethernet cards in both PCs and also a Gigabit Ethernet hub on the wall?
Or just a gigabit network switch between them is ok?

EricBoiseLGSVL, 2021-03-25:
I recommend as fast as possible to facilitate fast data transfer.  It will work with just the gigabit hub but you might have issues processing the data and it could effect performance.

closed 2021-07-06 by lemketron