lgsvl/simulator #891: Problems about LiDAR with high rotation frequency

FanLu97, 2020-08-02:
Hi,
Have you performed experiments on the simulator when LiDAR rotation frequency is about 50Hz?
And do you have recommended system specs if the LiDAR rotation frequency is set to 50~100Hz?
Thanks.

EricBoiseLGSVL, 2020-08-03:
@FanLu97 What hardware are you running now?

FanLu97, 2020-08-04:
> @FanLu97 What hardware are you running now?

Intel core i7-9700K (8 cores, 3.6Hz) and RTX 2070 Super with 16 GB RAM.

EricBoiseLGSVL, 2020-08-04:
@FanLu97 That seems like excellent hardware.  Are you running a AD stack on the same machine?  What sensors are you running with the Lidar?  What version of simulator are you using?

FanLu97, 2020-08-04:
> @FanLu97 That seems like excellent hardware. Are you running a AD stack on the same machine? What sensors are you running with the Lidar? What version of simulator are you using?

I just run ROS1 on the same machine (Only for visualization and data collection). Others sensors: GPS, GPS Odometry, IMU and vehicle control. The version of simulator is 2020.05. The simulator runs well when the LiDAR rotation frequency is 10Hz. When I modify the rotation frequency to 40 or 50Hz, the simulator is obviously not smooth, and the point cloud is not updated at the given frequency.

EricBoiseLGSVL, 2020-08-04:
@FanLu97 This may be a bug.  I'll have our team look into it

martins-mozeiko, 2020-08-04:
I'd say this is expected and not a bug. To have lidar operate smoothly and in real-time rendering performance needs to be something like 4x to 5x the lidar rotation frequency. So for 10Hz it will work great if your FPS is >50. For 100Hz your FPS needs to be >500 for it to work well.

And this is only rendering part. ROS1 bridge uses very inefficient protocol. It will not work well for higher beam count or rotation frequency. It will use too much CPU and too much bandwidth. Something like 16 beam lidar probably will max out bridge for 20-30Hz (haven't done actual measurements, that's just my guess). We have ROS2 bridge that should work better - as in use significantly less bandwidth, but that requires ROS2: [ros2-lgsvl-bridge](https://github.com/lgsvl/ros2-lgsvl-bridge). But it will still be limiting for high data rates because it will use too much CPU. To fix that you probably need to use directly velodyne lidar UDP based communication and skip ROS: https://www.lgsvlsimulator.com/docs/lidar-plugin/ Not sure about performance characteristics of this one.

Anyways, I would first figure out if you can get something like 400 to 500 rendering framerate for your needed 100Hz rotation frequency. Without that, lidar simulation won't be smooth.