lgsvl/simulator #892: Question: Usage of CAN bus data of LGSVL

ashwinsarvesh, 2020-08-02:
Hello,

I have a question.
I want to pack the CAN data of the simulator into CAN messages and send it to a CAN bus hardware connected to the PC.

Can I directly use the below variables in this file RosConversions.cs[](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/RosConversions.cs),  to represent the data field of my can message? 

- data.Speed (Line 259)
- data.LeftTurnSignal (Line 268)
- data.RightTurnSignal (Line 269)

Or is it **only** possible to use this data by subscribing to the ROS topic /apollo/canbus/chassis? If other ways are available to access this data please tell me. Kindly help.

EricBoiseLGSVL, 2020-08-03:
@ashwinsarvesh You could create UDP messages to the driver of the CAN Bus hardware.  Look at our UDP Lidar sensor as an example

ashwinsarvesh, 2020-08-04:
Thank you for your answer.
That means there are only 2 ways available to send the sensor data. One is via bridge and the other is via UDP packets. Am I right?

To send via UDP, I need to create a can bus sensor plugin and then use Ethernet to CAN converter to get the can messages on the can bus.
Am I right?

EricBoiseLGSVL, 2020-08-04:
@ashwinsarvesh We have two types of communication with examples currently.  Yes, we worked directly with the driver to communicate

ijklmnopqr, 2022-04-06:
@ashwinsarvesh was you able to get data from CAN sensor and send it via PCAN?