lgsvl/simulator #894: how to modify HD maps for Apollo

AlexandrZabolotny, 2020-08-03:
Hello team!  
Help me understand how to create HD maps  

I want to modify, for example, "autonomous_stuff" for Apollo  
I follow the instructions  
https://www.lgsvlsimulator.com/docs/map-annotation/#import-map-annotations  

But when I load the "base_map.bin" file, I get the error  
[image]

luqiang21, 2020-08-03:
@AlexandrZabolotny You have already imported the map named: `Map_LGSVL_0`. If you want to import it again, you need to either rename this one, or delete it. If you want to see the map annotations, you need to open `Simulator` -> `Map Annotation` window, then select `View ALL`.

AlexandrZabolotny, 2020-08-03:
Thank you for your quick response.
Now all OK

AlexandrZabolotny, 2020-08-06:
Sorry for my persistence. But I was unable to add a new mod (parking space) to the imported project.
According to your recommendation, I have uploaded the map (autonomous_stuff / base_map.bin). By the way, I can't see the parking spaces.

I now want to add parking spaces to the map according to the instructions
https://www.lgsvlsimulator.com/docs/map-annotation/#create-parking-space

But I cant see TARGET_WAYPOINT

[image]
[image]

Perhaps you have any idea what the problem is?

luqiang21, 2020-08-06:
@AlexandrZabolotny To annotate, you need to have a plane or road with mesh collider on it as described [here](https://www.lgsvlsimulator.com/docs/map-annotation/#creating-a-new-map). Then you can see the TARGET_WAYPOINT. 

Also, for AutonomouStuff parking lot, did you check our content website? [link](https://content.lgsvlsimulator.com/maps/autonomoustuff/) I believe we already have parking space annotated in the map.

AlexandrZabolotny, 2020-08-10:
First of all, I want to thank you for a very useful product.

> Also, for AutonomouStuff parking lot, did you check our content website? link I believe we already have parking space annotated in the map.  
I used your site data. But why can't I see the parking space when importing base_map.bin according to [the instructions](https://www.lgsvlsimulator.com/docs/map-annotation/#import-map-annotations) ?  I want to run [the open space algorithm](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/Open_Space_Planner.md).  But the algorithm does not work for me. I don't know the reason, so I am trying to modify the HD-map. It is also interesting to understand how your HD-map constructor works.  

> @AlexandrZabolotny To annotate, you need to have a plane or road with mesh collider on it as described here. Then you can see the TARGET_WAYPOINT.  
1. Can I create a plan or a road with mesh collider for AutonomouStuff myself in order to modify the HD-map ?.
2. Is it possible in the future will use your [constructor](https://www.lgsvlsimulator.com/docs/map-annotation/#creating-a-new-map) to create an HD-map when using Apollo on real cars? If so, how can I get(create) a plan or road with mesh collider? I have local_map which I got according to this [link](https://github.com/ApolloAuto/apollo/blob/master/scripts/msf_simple_map_creator.sh).

luqiang21, 2020-08-12:
@AlexandrZabolotny Sorry I noticed that our apollo and lanelet2 exporters support parking spaces. But importers don't support parking spaces yet. You need to either annotate manually or modify current apollo or lanelet2 importers to import parking spaces.

1. Yes, you can create a plane with mesh collider and modify the imported HD-map yourself.
2. We used our map for real car testing, it matches the real world, and we were able to run Apollo-3.5. For plane image, you can stitch the lossy maps you generated and then set correct `MapOrigin`.

AlexandrZabolotny, 2020-08-12:
1. I could not find on your site how to create a plane with a mesh collider? Can you give a link to instructions for this case?

2. I want to clarify. Do I understand correctly?
A lossy map has several folders with images of some areas. As shown [here](https://github.com/ApolloAuto/apollo/blob/master/docs/howto/images/msf_localization/map_node_image.png)
I have to stitch all the images together so that it is one image. Am I right?
If so how can I use this image to create the plane with the mesh collider?

EricBoiseLGSVL, 2020-08-12:
@AlexandrZabolotny 
1) To create a plane in Unity
- Select GameObject -> 3D Object -> Plane
- Right click in the External/Environments/YourNewMapFolder/Models/Materials/ and select Create -> Material
- Name Ground.mat
- In External/Environments/YourNewMapFolder/Models/Materials/ import the Texture that you are going to apply to the new Material.
- Select the Ground.mat and in the inspector drag the Texture to the BaseMap slot.
- Select the Plane mesh and drag the Ground.mat to the Materials[0] slot
- Drag the Plane (Rename it first) into External/Environments/YourNewMapFolder/Prefabs/ folder
- Please look at some Unity development tutorials online, these will help immensely when creating your new map if you are not familiar with this engine

The Plane will have a MeshCollider automatically added.  Mess with the mesh Scale and material UV Offset to fit.

2) Yes, I recommend Photoshop and create an image with the combined size and add each piece to the larger image, then save and use that as your materials BaseMap texture

luqiang21, 2020-08-12:
@EricBoiseLGSVL Thank you for answering.
@AlexandrZabolotny I just found the script I used before to stitch multiple images from lossy map folder. Hope it can be useful for you. https://github.com/lgsvl/apollo-3.0/blob/simulator/scripts/combine_pngs_from_pcd.py

AlexandrZabolotny, 2020-08-18:
@luqiang21 , @EricBoiseLGSVL 
I managed to create a map following your recommendations!
Thank's

closed 2020-08-18 by AlexandrZabolotny