lgsvl/simulator #895: Reading sensor data with Apollo via Cyber

VentiOvaltine, 2020-08-03:
I am running the LGSVL simulator with Apollo 5.5 and would like to read GPS/Lidar sensor data by using CyberRT with python.

After trying the tests under apollo/cyber/python, every python files imports a module "_cyber_wrapper" which does not exist under any folder.
[image]

Are there any suggestions on how to read data and how to run the cyber tests successfully? Thank you very much!

EricBoiseLGSVL, 2020-08-03:
@rongguodong Any idea on this?

rongguodong, 2020-08-03:
This does not sound like a simulator question. Are you trying to run some scripts provided by Apollo and get errors? If so, maybe you want to ask at Apollo's GitHub?

lemketron, 2020-08-04:
You also asked about this in #493, and [this comment](https://github.com/lgsvl/simulator/issues/493#issuecomment-667940122)suggested it might be an issue with cyber_py in apollo master. Please file an issue in the [apollo github issues](https://github.com/ApolloAuto/apollo/issues/). Feel free to link back here (or reference this issue in your apollo issue) so that others may locate the eventual solution.

VentiOvaltine, 2020-08-04:
This is the issue in Apollo. https://github.com/ApolloAuto/apollo/issues/12023

closed 2020-10-20 by EricBoiseLGSVL