lgsvl/simulator #896: Platform StandaloneLinux64 with device OpenGLCore is not supported with High Definition Render Pipeline

yumianhuli2, 2020-08-04:
Platform StandaloneLinux64 with device OpenGLCore is not supported with High Definition Render Pipeline, no rendering will occur
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr) (at /home/builduser/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:187)

Then I pressed the open browser button, The browser page is blank.

EricBoiseLGSVL, 2020-08-04:
@yumianhuli2 No, Linux requires Vulkan

yumianhuli2, 2020-08-05:
> @yumianhuli2 No, Linux requires Vulkan

Hello! I typed
1、sudo apt install vulkan-utils

2、sudo apt install libvulkan1 mesa-vulkan-drivers vulkan-utils

3、vulkaninfo

ERROR: [Loader Message] Code 0 : libGLX_nvidia.so.0: cannot open shared object file: No such file or directory
ERROR at /build/vulkan-tools-1.2.135.0~rc2/vulkaninfo/vulkaninfo.h:240:vkEnumerateInstanceExtensionProperties failed with ERROR_INITIALIZATION_FAILED

My system is ubuntu18.04
My driver is :
nvidia-smi
Wed Aug  5 11:16:24 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.57       Driver Version: 450.57       CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Quadro P5000        Off  | 00000000:21:00.0  On |                  Off |
| 26%   40C    P8     8W / 180W |    133MiB / 16270MiB |     27%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2182      G   /usr/lib/xorg/Xorg                131MiB |
+-----------------------------------------------------------------------------+

EricBoiseLGSVL, 2020-08-06:
You can use Nvidia control panel to force the application to use the correct GPU.  You might need to remove old drivers and try a clean install

closed 2020-08-08 by yumianhuli2

Addi2020, 2020-12-30:
@EricBoiseLGSVL  I am also facing the same issue, how to solve it? I reinstall NVIDIA drivers 440-100.
nvidia-smi (result inside and outside docker)
Wed Dec 30 13:36:22 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.100      Driver Version: 440.100      CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 166...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   42C    P8     2W /  N/A |     84MiB /  5944MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      5659      C   ...o/bazel-bin/modules/dreamview/dreamview    73MiB |
+-----------------------------------------------------------------------------+

I selected nvidia as a primary card

prime-select query
nvidia
apollo5.5$ sudo prime-select nvidia
Info: the nvidia profile is already set

but the picture below donot show nvidIia graghic card.

[image]

EricBoiseLGSVL, 2020-12-30:
Try this, hope is solves the issue.
[image]