lgsvl/simulator #897: Still needs put  assets in Assets/External/Environments？

yumianhuli2, 2020-08-04:
Still needs put  assets(Vehicle/map) in Assets/External/Environments？And where is the download-urls now?
And how are these external assets related to and different from assetBundle resources downloaded from the browser？
Thanks！

EricBoiseLGSVL, 2020-08-04:
@yumianhuli2 Yes, External assets are for asset creation in the source. AssetBundles and AppData are for loading a simulation at runtime and if running Loader.scene in Unity Editor

yumianhuli2, 2020-08-05:
> @yumianhuli2 Yes, External assets are for asset creation in the source. AssetBundles and AppData are for loading a simulation at runtime and if running Loader.scene in Unity Editor

Thanks! But what's the difference between  two assets？

EricBoiseLGSVL, 2020-08-05:
@yumianhuli2 
- Simulator/Assets/External/ is cloned repos source files for building your own asset bundles
- Simulator/AssetBundles/ are the built asset bundles
- AppData/ is the downloaded asset bundles

- This is environments/vehicles/npcs/controllables/sensors not just mesh data
- Delete your data.db and backup.db in AppData LG Simulator folder and run the 05 release build.  Then open browser and WebUI will download all default assets.  You will need to clone the repos into External yourself

closed 2020-08-08 by yumianhuli2