lgsvl/simulator #898: How to locate the object around the ego vehicle?

ForresterLu, 2020-08-04:
Hello team,

Recently, I am working on Apollo 5.0 and I try to obtain the location(like: x, y, z) of the objects around the ego vehicle, the objects can be NPC vehicle, pedestrain, and so on. I can obtain the position of the object only when the object is added by calling API, but how to obtain object position when the objects are randomly added. 
Is there any way for me to obtain the position in that case?

lemketron, 2020-08-04:
Random NPC info may not be available through the API, but you should be able to see it in the 2D and 3D ground truth sensor data if you add a ground truth sensor and connect to the configured sensor bridge.

ForresterLu, 2020-08-05:
> @ForresterLu This is not available currently. We will discuss internally and see if we could add it to the roadmap. In the meantime, you can alter the API code in simulator and in Python API to achieve this functionality.

Do you mean I need to modify the source code of Python API and API code in simulator?

ForresterLu, 2020-08-05:
> Random NPC info may not be available through the API, but you should be able to see it in the 2D and 3D ground truth sensor data if you add a ground truth sensor and connect to the configured sensor bridge.

Thanks you @lemketron , could you please explain more detail of how to do this? Better with an example to show how to obtain position of object on runtime.

EricBoiseLGSVL, 2020-08-05:
@ForresterLu Groundtruth data is sent over bridge, subscribe to this data with Ros, Ros2, cyber.