lgsvl/simulator #900: Agents still exits after sim.reset() is called

ForresterLu, 2020-08-05:
Hello team,

I loaded  `SanFrancisco` as my running environment in LGSVL and added some agents like NPC vehicle and pedestrain to the environment. But when I want to reset the evnironment by calling `sim.reset()`, I found the agents (NPC and pedestrains) still in my new reseted environment, seems this is an bugs need to be fixed in LGSVL?

ntutangyun, 2020-08-06:
sounds like a bug. 

If you use the PythonAPI to add those agents, I think the `sim.reset()` should be able to remove them all. 

what you could try is when you add the agent, you'll get a `UID` in response. You could use the `removeAgent` API to remove those agents one by one in the new environment.

ForresterLu, 2020-08-06:
> sounds like a bug.
> 
> If you use the PythonAPI to add those agents, I think the `sim.reset()` should be able to remove them all.
> 
> what you could try is when you add the agent, you'll get a `UID` in response. You could use the `removeAgent` API to remove those agents one by one in the new environment.

OK, thank you, that's exactly what I deal with old agents for now.

closed 2020-08-28 by EricBoiseLGSVL