lgsvl/simulator #906: Can I build the source code into WebGL platform?

gyp199009, 2020-08-07:
Hello, 
Since the source code can be built for either Windows, Linux or Mac, and meanwhile unity supports WebGL platform,  I am wandering is it possible to build the source code for WebGL ? If not, it there a way to access the simulator through a web page?
Thank you so much!

ntutangyun, 2020-08-07:
haha, interesting. have you tried yet?

EricBoiseLGSVL, 2020-08-07:
@gyp199009 No sorry, WebGL won't support HDRP.  We are currently looking at other ways for users to access the simulator.  We will have more information soon :)

rongguodong, 2020-08-07:
According to Unity's [official document](https://docs.unity3d.com/Packages/com.unity.render-pipelines.high-definition@7.1/manual/index.html?_ga=2.64773291.549701310.1596816638-434578097.1596665788), HDRP does not OpenGL or OpenGL ES, and WebGL is based on OpenGL ES. So the current HDRP-based simulator is not compatible with current WebGL.

However, there are some discussions within Khronos and W3C to extend WebGL to support Vulkan or have WebGPU to support Vulkan. Once Vulkan is supported, Unity may extend its WebAssembly support to support Vulkan, and thus support HDRP via web.

yumianhuli2, 2020-08-08:
> According to Unity's [official document](https://docs.unity3d.com/Packages/com.unity.render-pipelines.high-definition@7.1/manual/index.html?_ga=2.64773291.549701310.1596816638-434578097.1596665788), [...]

it seems like a long story:)

EricBoiseLGSVL, 2020-08-10:
@yumianhuli2 Yes, I would not hold my breath for Unity to do any extra work with WebGL.  It has been neglected for a long time.

yumianhuli2, 2020-08-11:
> @yumianhuli2 Yes, I would not hold my breath for Unity to do any extra work with WebGL. It has been neglected for a long time.

:)

closed 2020-08-14 by EricBoiseLGSVL