lgsvl/simulator #907: A不呕吐

yumianhuli2, 2020-08-08:


ntutangyun, 2020-08-10:
You could ask this question in [Apollo github](https://github.com/ApolloAuto/apollo).

> I have a custom data structure. How to use Cyber to create and use it?

I think the basic steps are
* create a `*.proto` file for your custom data structure under Apollo's build directory
* compile the proto file with `protoc`, target language is c#. i.e. csharp.
* copy the compiled the file into corresponding directory inside the simulator:  `simulator/Assets/Scripts/Bridge/Cyber/`
* then you should be able to use the compiled class in your unity script. 

> If it is an empty scenario and I need to use Cyber to communicate with an external program, what is the least script I need to have?

As far as I know, LGSVL uses WebSocket to communicate with outside programs. e.g. the API endpoint and the cyber bridge. 
You could either make use of the existing API endpoint, or create new websocket classes for your self. 

> Is there a simplest example scenario for teaching? Where is the relevant introduction document?

What do you mean by example scenario?

EricBoiseLGSVL, 2020-08-12:
@yumianhuli2 Yes, @ntutangyun pretty much covered everything.  For 3.) we have created detailed video tutorials for Autoware and Apollo and hope to release them soon.

yumianhuli2, 2020-08-16:
> @yumianhuli2 Yes, @ntutangyun pretty much covered everything. For 3.) we have created detailed video tutorials for Autoware and Apollo and hope to release them soon.

I'm really looking forward to this！

closed 2020-08-16 by yumianhuli2