lgsvl/simulator #908: map annotation

yumianhuli2, 2020-08-08:
Using the map annotation can produce xodr and basemap. Bin. Then xodr for simulation, basemap. bin into the specified folder（https://github.com/ApolloAuto/apollo/tree/master/modules/map/data），Do I understand that right?

luqiang21, 2020-08-08:
@yumianhuli2 xodr is openDrive format, I am not sure what do you mean by simulation visualization. To run apollo, yes, you need Apollo map: basemap.bin and place it in corresponding folder inside Apollo.

yumianhuli2, 2020-08-09:
> @yumianhuli2 xodr is openDrive format, I am not sure what do you mean by simulation visualization. To run apollo, yes, you need Apollo map: basemap.bin and place it in corresponding folder inside Apollo.

After the map annotation process can I export basemap.bin and XXX.xodr at the same time？

yumianhuli2, 2020-08-09:
> > @yumianhuli2 xodr is openDrive format, I am not sure what do you mean by simulation visualization. To run apollo, yes, you need Apollo map: basemap.bin and place it in corresponding folder inside Apollo.
> 
> After the map annotation process can I export both basemap.bin and XXX.xodr？

closed 2020-08-09 by yumianhuli2