lgsvl/simulator #909: How to move reverse/backwards and add another type of controller?

bobbyhaliwela, 2020-08-08:
I've played around a bit with lgsvl simulator 2019 a year ago, then just downloaded the 2020.05 version a few days ago. I noticed that there is no option to shift the gear to move backwards/reverse? If i want to add the option for this, what part of the code should i be looking into? Also, if i want to add another type of joypad controller like PS4/Xbox, what part of the code should i be looking into? Thanks.

bobbyhaliwela, 2020-08-08:
> I've played around a bit with lgsvl simulator 2019 a year ago, then just downloaded the 2020.05 version a few days ago. I noticed that there is no option to shift the gear to move backwards/reverse? [...]

Found the script, in order for us to shift gear, we can press 'page up' or page down' button. And it seems, we can use Xbox joystick, but what about PS4?

lemketron, 2020-08-08:
You can add a keyboard and/or wheel control to the sensor configuration as documented here: https://www.lgsvlsimulator.com/docs/sensor-json-options/#keyboard-control

The keyboard shortcuts are documented here: https://www.lgsvlsimulator.com/docs/keyboard-shortcuts/

Page up and page down do indeed shift gears; in previous versions of the simulator there were on-screen indicators to show if the parking brake was engaged and what gear the car was in but as you cannot see that now you should be careful when toggling the brake or changing gears. 

Also, if you are in interactive mode, you'll need to press the "play" triangle before the car will move but once you do that you should be able to press the up arrow to make the car go forward.

EricBoiseLGSVL, 2020-08-12:
@bobbyhaliwela If you need specific controller support please look at Assets/Scripts/Controllers/SimulatorControls.
Here you can add support for any control type you need.  Then make a custom sensor for these new controls.  Look at WheelControl and KeyboardControl sensors as examples