lgsvl/simulator #910: Custom Pedestrian Path

gizegizem, 2020-08-10:
Hello, I am trying to make my own pedestrian path. I have followed the instructions at website. I just created the path. But it didn't work.  How can I make a connection between the path and the pedestrian itself. 

Can you help me? Thanks.

ntutangyun, 2020-08-10:
Have you tried this script ?

https://github.com/lgsvl/PythonAPI/blob/master/quickstart/16-pedestrian-follow-waypoints.py

EricBoiseLGSVL, 2020-08-10:
@gizegizem yes, to have a specific pedestrian walk a specific path, you should use the API.  If you want random pedestrians to walk paths then you will need to annotate the paths and make sure it is a child object of the MapAnnotation for the map.  You also need a map Nav Mesh.

- Create an empty game object and put all ped annotation under it like so.
[image]

- Then you need a Unity Nav mesh for the meshes in the scene.  Select Window -> AI -> Navigation
- Select meshes and mark them Navigation Static like so
[image]

- Then mark the types of nav mesh for each mesh like so
[image]

- Then bake the nav mesh and save the scene like so.  This will create a nav mesh object
[image]

Make sure the nav mesh covers the entire ground, we do a check at the spawn info object to make sure there is one present in the map and if it fails to raycast to it, it will disable the pedestrian feature.

gizegizem, 2020-08-12:
Thank you so much @EricBoiseLGSVL 🥇  , it did work.

closed 2020-08-12 by EricBoiseLGSVL