lgsvl/simulator #911: Custom Api command not working on binary

SeymurSm, 2020-08-11:
Hello,
I have added new command file in Scripts/Api/Command folder to add custom objects from PythonApi. The command works well when I start Api mode from Unity Editor. It is not working when the Api mode started from binary file. I have added the new command and build binary. It should work same as editor. Is it something to do after adding custom scrips to Scripts/Api/Command ?

ntutangyun, 2020-08-11:
usually, it should work. You could try to create a dummy command, like returning current time, and see it works at all?

It might be due to the part which adds the custom object.

EricBoiseLGSVL, 2020-08-12:
@SeymurSm Are you making a new binary from the altered source?  Did you clone and change the PythonAPI as well as the simulator source?